bblanchon/ArduinoJson #190: Use number with createnestedobject

sticilface, 2016-01-06:
is is possible to create a nested object, that consists of a number? 

JsonObject::createNestedObject(const char *key)

Get an error when it is not const char?

sticilface, 2016-01-06:
I guess you just use array!

bblanchon, 2016-01-06:
`JsonVariant` is a type that can contain several kind of value: string, number, etc.

```c++
JsonVariant v;
v = "hi!";
v = 42;
```

`JsonArray` and `JsonObject` are collections of `JsonVariant`.

But I don't understand your question... could you elaborate?

sticilface, 2016-01-06:
I'm being a spanner.  Thanks for the reply.  but please ignore.

closed 2016-01-06 by sticilface

sticilface, 2016-01-06:
Actually... maybe I'm not being a spanner... here is a sample JSON.

```
{
  "0": {
    "name": "SimpleColor",
    "brightness": 177,
    "speed": 1950289429,
    "color1": {
      "R": 184,
      "G": 33,
      "B": 64
    }
  }
}
```

Here the "0":  which is a createNestedObject("0") must be a const char *, but is there a reason it cannot be an integer?

bblanchon, 2016-01-07:
You mean, like this:

```json
{
  0: {
    "name": "SimpleColor",
    "brightness": 177,
    "speed": 1950289429,
    "color1": {
      "R": 184,
      "G": 33,
      "B": 64
    }
  }
}
```

It's not valid JSON anymore, see [json.org](http://json.org]):

[image]
[image]

sticilface, 2016-01-07:
spoon accepted back!  
Thanks