bblanchon/ArduinoJson #192: Size

donnib, 2016-01-07:
Hi,
I guess this is an easy question. I have 512bytes in the ESP8266 RTC where i would like to save 10 sets of Temperature and Humidity and i am not sure i understand how to calculate exactly :

Object size on ESP8266

Type	Size
JsonVariant	8
JsonArray of N element	8 + 12 * N
JsonObject of N element	8 + 16 * N

So i structure the data like this :

[
    {"T":"20","H":"40"},
    {"T":"20","H":"40"},
    ...
]

So the buffer get's to be :

const int BUFFER_SIZE_R = JSON_ARRAY_SIZE(10) + JSON_OBJECT_SIZE(20);

Would this fit inside 512bytes ? As far as i can calculate it does, 8+12*10=128, 8+16*20=328 so in total 456bytes, is this totally misunderstood ? I guess that if it doesn't fit i could save them as a byte array then serialize it into JSON then send it over the network.

bblanchon, 2016-01-07:
Hi, 

If you have an array containing 10 objects of 2 elements, then the formula is:

```c++
const int BUFFER_SIZE_R = JSON_ARRAY_SIZE(10) + 10*JSON_OBJECT_SIZE(2);
```

But this is not the size of the JSON string, it's the memory required to store the object tree in memory (ie the size of the `StaticJsonBuffer`).

The size of the JSON string cannot be known at compile time, as is depends of the values inserted at run time.
You can know measure the size of the JSON string by calling `JsonArray::measureLength()`.

donnib, 2016-01-07:
Ok so my buffer size was wrong, i guess i did not run it to fill it up because it probably would have failed.

Regarding the size, i should be able to calculate it since i know it's numbers and what kind of numbers it is no ? Any idea how to calculate ?

bblanchon, 2016-01-07:
Easy trick: you type the whole JSON in a text file and you look at the size of the file.

donnib, 2016-01-07:
yes of course, 201bytes so it can fit ;)

closed 2016-01-10 by donnib