bblanchon/ArduinoJson #193: Is there a way to gracefully not find a value?

sticilface, 2016-01-07:
On the ESP8266, when parsing JSON, if the key is not present I get an exception and the device crashes.  I get around it my wrapping every assignment with containsKey().  Is it possible to have the assignment operator fail gracefully if the key is not present.  Or is this intended behaviour? 

(sorry for posting so many queries, I'm using you lib extensively, and it is great)

Thanks

bblanchon, 2016-01-07:
ArduinoJson should not make the device crash.
Can you please copy/paste a small code snippet?

sticilface, 2016-01-07:
It occurs when ever I try to read a key that is not there... 
crashes if there is no key "brightness"
```
_brightness = (uint8_t)current["brightness"].as<long>(); 
```
Doesn't crash as it checks if there is a "brightness" key first
```
if (current.containsKey("brightness")) { 
_brightness = (uint8_t)current["brightness"].as<long>(); 
}
```

sticilface, 2016-01-07:
Its done this for ages.  So i just developed the habit of using containsKey.  But.. i just spent an hour trying to find a bug with comes fairly complex code, and it turned out to be this, as i forgot to wrap my assignment.

bblanchon, 2016-01-07:
It's supposed to return 0.
Well, this is a true bug.

sticilface, 2016-01-07:
It doesn't have to be int, happens with const char *, jsonObject..  all of them i think

bblanchon, 2016-01-07:
BTW, you can replace:

```c++
_brightness = (uint8_t)current["brightness"].as<long>(); 
```

by

```c++
_brightness = current["brightness"]; 
```

sticilface, 2016-01-07:
Cool thanks..  not sure why i did that.. think i was experimenting with the explicit casting stuff, as i was reading the wiki at the time:)

bblanchon, 2016-01-07:
I'll fix this as soon as I'm back home, in 4 to 5 hours.

sticilface, 2016-01-07:
No problem at all!  

No urgency either!   Thanks for all the hard work!

bblanchon, 2016-01-07:
I tried to reproduce the bug on my PC, but it works.

I'm now trying on a SparkFun ESP8266 and it's not crashing.
I started from the JsonParserExample and changed the keys, so that the don't match.
Can you help my find this nasty bug ?

sticilface, 2016-01-08:
I'll agree with you. Editing the example did not reproduce it.  I'm stumped. 
I know that if i try to use a value that is not there.  say strcmp, i get an exception, as the const char * returned by arduinojson is a null pointer.  so that makes sense.  BUT.. i was pretty sure i was getting these problems at the point of assignment!!!  I will try and create it!

bblanchon, 2016-01-13:
So... did you saw the [Bug Foot](http://fr.urbandictionary.com/define.php?term=bugfoot) again? :wink:

sticilface, 2016-01-14:
nope!  but i've not had much free time.  I'm suspecting it is to do with parsing a nullptr out of arduinojson and not catching it.  wrapping everything in containsKey... would obviously stop it..  I swear that i thought of that and it was still doing it.  anyway... I'll keep my eyes open!

bblanchon, 2016-01-14:
OK, let's keep this issue open for a while.
Maybe, some other user will add a comment.

fishix10, 2016-01-19:
Not sure if this helps, but my [terribly written] program has this bug too. When I parse the string, it returns success. But then the tree wouldn't have the structure I expect and when I try to access it I'd get a reset. My json is quite complex too... so now I'm left with something like this, basically checking at all levels:

    if ((*root).containsKey("query")) {
      if (((JsonObject&)((*root)["query"])).containsKey("results")) {
        if (((JsonObject&)((*root)["query"]["results"])).containsKey("results")) {
          if ((*root)["query"]["results"]["results"].is<JsonArray&>()) {
            if ((*root)["query"]["results"]["results"][r].is<JsonObject&>()) {
              const char* tmp = (*root)["query"]["results"]["results"][r]["a"];
              Serial.println(tmp);
              return;
            } else Serial.println("noo 1");
          } else Serial.println("noo 2");
        } else Serial.println("noo 3");
      } else Serial.println("noo 4");
    } else Serial.println("noo 5");

I'm sure there's a better way to do this and I'm not even sure I'm doing this right, but at least I don't get random resets anymore. Any suggestions would really help :(

bblanchon, 2016-01-19:
@fishix10, I designed the library so that you should be able to write this:

```c++
const char* tmp = (*root)["query"]["results"]["results"][r]["a"];

if (tmp) {
    Serial.println(tmp);
} else {
    Serial.println("Error: .query.results.results[r].a not found!");
}
```

Even if any of the members in the path to `.query.results.results[r].a`, the program should not crash.
Instead it should simply propagate the `NULL` value.

Please tell me if this doesn't work properly.

fishix10, 2016-01-19:
Thanks @bblanchon! I think I've tried that before, but it didn't work. I just fried my ESP8266 today (sad). I'll try it again and let you know after I get a new unit tomorrow.

bblanchon, 2016-01-23:
Guys, I'm closing this issue because [people starts to beleive that there is a bug](https://github.com/tzapu/WiFiManager/issues/70#issuecomment-172530256) although there is currently no evidence of it.

If you experience a crash that is caused by ArduinoJson, please open a new issue and I'll treat it with the highest priority.

Let's make things clear: ArduinoJson should not make the device crash!
The only acceptable situation for a crash is when you use a `StaticJsonBuffer` that is bigger than the available size on the stack, because this is impossible to detect.

However, calling `strcmp()`, `strcpy()`... with a `NULL` parameter will make the device crash.
But it's the responsibility of your program to check that the value is not `NULL`.

Lastly, calling `JsonObject::containsKey()` is often not necessary.
It sufficient to check that the value is not `NULL`.

Thanks for reporting this anyway, and thanks for using ArduinoJson.

closed 2016-01-23 by bblanchon