bblanchon/ArduinoJson #194: Reading file from SPIFFS, or stream?

sticilface, 2016-01-07:
I'm doing a fair amount of arduinoJsoning from a SPIFFS file.  
At the moment I have to copy the whole lot into a buffer for use with arduinojson, but I'm wondering if this is really necessary.  

The file is seekable, writeable etc... is there any methods you can think of that i could use to parse it directly to ardunojson... 

if i wrote a class that dealt with the underlying spiffs, returned a pointer to char and had operators such as [], would that conceivably work? 
```
	jsonObject & root = jsonBuffer.parseObject( myspecialfunc() );
```
?

bblanchon, 2016-01-07:
Unfortunately, ArduinoJson need to have the whole JSON string in memory.
That's the fundamental design decision that allows the library to work with zero copy and zero malloc.

sticilface, 2016-01-08:
Fair enough.

closed 2016-01-08 by sticilface