bblanchon/ArduinoJson #195: Newbie: Common base object for JsonArray and JsonObject?

andig, 2016-01-08:
I have a couple of functions that send json responses on ESP8266. The recurring pattern in by newbies code:

    char buffer[512];
    json.printTo(buffer, sizeof(buffer));
    g_server.send(200, "application/json", buffer);

I would like to make this a function- however the json objects are either `JsonArray` or `JsonObject` that- as far as I can see do not have a common ancestor. Is this correct?

bblanchon, 2016-01-08:
Hi Andreas,

This is a very good question and I'm pretty sure it has never been asked before.

You're right: there is no base class for `JsonArray` and `JsonObject`.
(Back in version 3.0, they used to derive from `Printable`, but I removed this inheritance to reduce the memory footprint.)

However, both `JsonArray` and `JsonObject` can be "stored" in a `JsonVariant`.
I put "stored" in quotes because the `JsonVariant` only contains a reference, not a copy.

So here is your function:

```c++
void sendJson(JsonVariant json) {
    char buffer[512];
    json.printTo(buffer, sizeof(buffer));
    g_server.send(200, "application/json", buffer);
}
```

andig, 2016-01-10:
I'm sorry, he ArduinoJson C++ is way above my head. I couldn't find the JsonVariant constructor in the sources- so how would I create one? Would be great if you could add the calling code as well. Much appreciated!

bblanchon, 2016-01-10:
```c++
JsonObject& obj = jsonBuffer.createObject();
obj["hello"] = "world";
sendJson(obj);
```

andig, 2016-01-10:
No common ancestor yet working as paramter type. You have to love C++ ;) thank you so much and thanks for bearing with me through my ignorance.

bblanchon, 2016-01-10:
Don't blame yourself, this one was very hard to catch.
The documentation is the weak point of this library.
I'm doing my best, but if you think you can help, don't hesitate to edit the wiki.

bblanchon, 2016-01-13:
Added to the FAQ:
https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-write-a-function-that-works-with-both-jsonarray-and-jsonobject

closed 2016-01-13 by bblanchon