bblanchon/ArduinoJson #196: How to pass nested object as function parameter

Diaoul, 2016-01-10:
Here is my JSON: `{"global": {"pattern": "rainbow"}}`
Somewhere in main:
```
  if (root.containsKey("global")) {
    readGlobal(root["global"]);
  }
```
Now the dedicated function:
```
void readGlobal(JsonObject& data) {
  if (data.containsKey("pattern")) {
    [...]
  }
}
```

With this it works but RAM usage is crazy. I tried JsonVariant but it lacks the `containsKey` method. Can you help me finding a more efficient way of achieving this?

Thanks for your great lib.

Diaoul, 2016-01-10:
Forget the memory consumption stuff. I think it's related to something else. But still, what's the best way to pass a known nested object as parameter to a function?

bblanchon, 2016-01-10:
The best way to pass an object (nested or not) to a function is to have a parameter of type `JsonObject&` (or `const JsonObject&`).
That's exactly what you did.

Now, if you don't known at compile time if it's an array or an object, you must use a parameter of type `JsonVariant`.
But in that case, you loose some specificities of the `JsonObject` class.
In particular, you don't have the `containsKey()` method.
If you need this function, you must cast the `JsonVariant` back to a `JsonObject&`.

For instance:

```c++
void myFunction(JsonVariant variant)
{
    if (variant.is<JsonArray&>())
    {
        JsonArray& array = variant;
        // ...
    }
    else if (variant.is<JsonObject&>()) 
    {
        JsonObject& object = variant;
        // ...
    }
    else
    {
        // ...
    }
}

bblanchon, 2016-01-13:
Added to the FAQ:
https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-write-a-function-that-works-with-both-jsonarray-and-jsonobject

closed 2016-01-13 by bblanchon