bblanchon/ArduinoJson #197: I have a string in a String type variable

mmiscool, 2016-01-10:
Hello,

I am trying to use a string in stead of hard coding the names of json files i want to extract. Having some significant difficulties.

excerpt of code:

    StaticJsonBuffer<200> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(Param0);

    if (!root.success()) {
      Serial.println("parseObject() failed");

    }
    else
    {

    int str_len = Param1.length() + 1;
    char AsAChar[str_len];
    Param1.toCharArray(AsAChar, str_len);
    MyOut = root[AsAChar];
    }

bblanchon, 2016-01-10:
Hi,

Assuming that `Param1` is a `String`, this should work:

```c++
MyOut = root[Param1];
```

mmiscool, 2016-01-11:
I get the following error when compiling now.

mmiscool, 2016-01-11:
Functions_and_var_management.ino: In function 'String GetMeThatVar(String)':
Functions_and_var_management:152: error: ambiguous overload for 'operator=' (operand types are 'String' and 'ArduinoJson::JsonObjectSubscript<const String&>')
Functions_and_var_management.ino:152:15: note: candidates are:
In file included from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../Arduino/String.hpp:16:0,
                 from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:13,
                 from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:9,
                 from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/DynamicJsonBuffer.hpp:9,
                 from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson.h:7,
                 from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:13,
                 from ESP8266Basic.ino:29:
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc1\cores\esp8266/WString.h:85:18: note: String& String::operator=(const String&)
         String & operator =(const String &rhs);
                  ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc1\cores\esp8266/WString.h:86:18: note: String& String::operator=(const char*)
         String & operator =(const char *cstr);
                  ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc1\cores\esp8266/WString.h:87:18: note: String& String::operator=(const __FlashStringHelper*) <near match>
         String & operator = (const __FlashStringHelper *str);
                  ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc1\cores\esp8266/WString.h:87:18: note:   no known conversion for argument 1 from 'ArduinoJson::JsonObjectSubscript<const String&>' to 'const __FlashStringHelper*'
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc1\cores\esp8266/WString.h:89:18: note: String& String::operator=(String&&)
         String & operator =(String &&rval);
                  ^
Functions_and_var_management:156: error: ambiguous overload for 'operator=' (operand types are 'String' and 'const ArduinoJson::JsonArraySubscript')
Functions_and_var_management.ino:156:15: note: candidates are:
In file included from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../Arduino/String.hpp:16:0,
                 from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:13,
                 from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:9,
                 from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/DynamicJsonBuffer.hpp:9,
                 from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson.h:7,
                 from C:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:13,
                 from ESP8266Basic.ino:29:
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc1\cores\esp8266/WString.h:85:18: note: String& String::operator=(const String&)
         String & operator =(const String &rhs);
                  ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc1\cores\esp8266/WString.h:86:18: note: String& String::operator=(const char*)
         String & operator =(const char *cstr);
                  ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc1\cores\esp8266/WString.h:87:18: note: String& String::operator=(const __FlashStringHelper*) <near match>
         String & operator = (const __FlashStringHelper *str);
                  ^
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc1\cores\esp8266/WString.h:87:18: note:   no known conversion for argument 1 from 'const ArduinoJson::JsonArraySubscript' to 'const __FlashStringHelper*'
C:\Users\user\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc1\cores\esp8266/WString.h:89:18: note: String& String::operator=(String&&)
         String & operator =(String &&rval);
                  ^
ambiguous overload for 'operator=' (operand types are 'String' and 'ArduinoJson::JsonObjectSubscript<const String&>')

bblanchon, 2016-01-11:
This is issue #118 and it's [covered in the FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-fix-error-ambiguous-overload-for-operator):

> There is a case where you need to help the compiler: it's when you convert a `JsonVariant` to a `String`.
>
>For example:
>
>```c++
>String ssid = network["ssid"];
>ssid = network["ssid"];
>```
>
>The first line will compile but the second will fail with the following error:
>
>```
>error: ambiguous overload for 'operator=' (operand types are 'String' and >'ArduinoJson::JsonObjectSubscript<const char*>')
>```
>
The solution is to remove the ambiguity with any of the following replacement:
>
>```c++
>ssid = (String)network["ssid"];
>ssid = (const char*)network["ssid"];
>ssid = network["ssid"].asString();
>ssid = network["ssid"].as<const char*>();
>ssid = network["ssid"].as<String>();
>```

closed 2016-01-13 by bblanchon