bblanchon/ArduinoJson #199: Header only library

bblanchon, 2016-01-11:
Arduino, PlatformIO, Stino, Energia... enough!
I'm tired of the packaging issues.
Let's go header only.

closed 2016-06-17 by bblanchon

DinoStray, 2017-12-29:
I wonder, there are so many files in src/ArduinoJson, how do you make them to only one header flie ? Which tool helps you, CMake?
I'd like you to  help me solve the problem, please

bblanchon, 2017-12-29:
Hi @DinoStray,

I use this horrible bash script:
https://github.com/bblanchon/ArduinoJson/blob/master/scripts/build-single-header.sh

It very slow, so I was planning on writing a C++ version.
Would you be interested, or is the script sufficient?

Regards,
Benoit

DinoStray, 2018-01-02:
I try to use the bash script, it's amazing.
I think the script is sufficient!