bblanchon/ArduinoJson #201: How to add json objects to a json array?

fmgomes, 2016-01-13:
Hi, is there any example on how to add json objects to a json array? From the API this seems possible, but I'm not getting how to do it correctly. I create the root object, then create an array on it like in the examples, but then how can I add json objects as the array elements?

Thanks for such a great library!!!

Sungurik, 2016-01-13:
See "Example 1"
https://github.com/bblanchon/ArduinoJson/wiki/Examples

bblanchon, 2016-01-13:
Hi @fmgomes,
From what I understood, you're looking for something like:

```c++
JsonObject& root = jsonBuffer.createObject();
JsonArray& array = root.createNestedArray("key");
JsonObject& object = array.createNestedObject();
```

Please copy/paste code here, so one can explain you what's wrong.

fmgomes, 2016-01-14:
@bblanchon, you are right, I have managed to make it work in the meantime, using the example refered by @Sungurik. Thanks to both of you! The code is the following:

    JsonObject& root = jsonBuffer.createObject();
    root["key1"] = 1000;
    JsonArray& array = root.createNestedArray("key2");

    JsonObject& object = array.createNestedObject();
    object["keyA"] = "AAA";
    object["keyB"] = "BBB";
    object["keyC"] = 111;

This introduces correctly the JSON object in the array and I am able to get the result i was looking for. Now an additional doubt, if I want to add several elements to the array (other objects like the above), for example in a loop, what is the correct way to do it? Is it possible to reuse the variable 'object'? If I try to add a second object to the array reusing the object variable (reference) used before, I get a compilation error (if I add the following code to the code above):

    object = array.createNestedObject();
    object["keyA"] = "ZZZ";
    object["keyB"] = "XXX";
    object["keyC"] = 999;

In this case I get the following error:

error: use of deleted function 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)'

Thanks again for your help!

Best regards

Fernando

bblanchon, 2016-01-14:
Hi Fernando,
Indeed, C++ doesn't allow to change what a reference points to.
Why don't you make `object` a local variable of the loop:

```c++
for (...) {
  JsonObject& object = array.createNestedObject();
  object["keyA"] = "ZZZ";
  object["keyB"] = "XXX";
  object["keyC"] = 999;
}
```

closed 2016-01-19 by bblanchon

bhattvishal, 2017-01-13:
What am I doing wrong here?

`for (...) {
		JsonObject& sensorChannel = realtimechannels.createNestedObject();
		sensorChannel["c"] = data[i].Id;
		sensorChannel["cn"] = data[i].Name;
		sensorChannel["v"] = data[i].Value;
		sensorChannel["t"] = GetISODateTime();
		sensorChannel["t_ms"] = data[i].TimeInMiliseconds;

		//Serial.println("Light is " + data[i].Name);
		//Serial.println("Light is " + data[i].Value);
		//realtimechannels.add(tempChannel);
	}`
But the json output does not contain all the channel values. See the output below
`"realtimes":[{"c":1,"cn":"Temperature","v":22.06,"t":"2017-01-13T09:08:12","t_ms":0},{"c":2}]}]`

bblanchon, 2017-01-13:
@bhattvishal Your `JsonBuffer` is probably too small.

See:

* [FAQ: Why some parts are missing?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-some-parts-are-missing)
* [ArduinoJson Asssistant](https://bblanchon.github.io/ArduinoJson/)

treii28, 2017-10-17:
Is it possible to return a JsonObject from another method and attach it elsewhere nested under an array?