bblanchon/ArduinoJson #202: Parsing array more than three elements get error.

sombatsombat, 2016-01-21:
With the Json with data array more than three elements:

{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.3,2.38,3.5,6.23]}

I got the parse error.

I'm using Sming framework for ESP8266. Don't know if this is a bug for here or not.

bblanchon, 2016-01-21:
I guess that you're using a `StaticJsonBuffer` that is too small to hold the object tree.

The ["Why parsing fails ?" of the FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails) will provide you with the necessary information.

bblanchon, 2016-01-21:
(Sorry, I clicked on "Close" accidentally.)

sombatsombat, 2016-01-21:
With the following code, the parseObject got success. But with the comment line uncommented (and the line above it is commented), the parseObject failed. 

    StaticJsonBuffer<200> jsonBuffer;
    
	char json[] =
     "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
      //"{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.3,2.38,3.5,6.23]}";

     JsonObject& root = jsonBuffer.parseObject(json);
	
	if (!root.success()) {
	    Serial.println("parseObject() failed");
	    return;
	}
	else{
	    root.prettyPrintTo(Serial);
    }

bblanchon, 2016-01-21:
Yes, that's exactly what I was saying.

Just increase the size of the `StaticJsonBuffer` or switch to `DynamicJsonBuffer`.

sombatsombat, 2016-01-21:
Increasing the size of StaticJsonBuffer from 200 to 300. The problem is gone.

At first I thought there might be a problem of your code because those two json strings have the same length. But apparently, it depends on other things too. Thank you.

bblanchon, 2016-01-21:
It's the size required to store the object tree in memory.
It entirely depends on the complexity of the JSON input

In the first case there were one object with 3 members, including an array of 2 elements.
In the second case, it's still one object with 3 members, but the array contains 4 elements.

You can compute, at compile time, the size required:

```c++
#define JSON_BUFFER_SIZE (JSON_OBJECT_SIZE(3) + JSON_ARRAY_SIZE(4))
```

However, since you seem a bit lost, I advice you to switch to a `DynamicJsonBuffer` because the size is automatic.

closed 2016-01-23 by bblanchon