bblanchon/ArduinoJson #203: Help with use this library

qblacksheep, 2016-01-22:
Hi, Could you help me to write program wchich using this library?
I need to:
1) i declare a json[] array
2) send a frist json[] array
3) recive id and replace this id with "undefined" id
4) send a new array with stored it
5) add a data from sensor to json array and send it 

I am using ws library
There is ws method where i want to do it:
[ard.txt](https://github.com/bblanchon/ArduinoJson/files/100116/ard.txt)
[image]
[image]

bblanchon, 2016-01-23:
I edited you question to add syntax highlighting.

It's pretty obvious that there is something wrong in the definition of `json`.

Then, in `root ["stored_id"] == id_device;`, it seems that you wanted to use `=` not `==`.

As a general programming advice, you should not ignore the compiler messages, I'm pretty sure it warn you about these...

qblacksheep, 2016-01-24:
Okay, thanks. I arleady have it. I can change id_device but i have another problem. I can change only first atribute. When i want change other value of atriubute its write it at the end of my json format. I want to change value of SendDataVoltage. 
Here is serial ss, and my code.
[image]

```javascript
/*
 * WebSocketClient.ino
 *
 *  Created on: 24.05.2015
 *
 */

#include <Arduino.h>
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <WebSocketsClient.h>

#include <Hash.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;

#define USE_SERIAL Serial
boolean  connected = true;
DynamicJsonBuffer jsonBuffer;
String SendDataVoltage =
"{\"action\":\"data\",\"signals\":[{\"interface_id\":\"int_1\",\"message\":{\"data_type\":\"voltage\",\"value\":\"0\"}}]}";

String SendDataLight =
"{\"action\":\"data\",\"signals\":[{\"interface_id\":\"int_0\",\"message\":{\"data_type\":\"light\",\"value\":\"5\"}}]}";
String registration = 
 "{\"stored_id\":\"undefined\"\,\"action\":\"register\",\"device\":{\"name\":\"sensor\",\"interfaces\":[{\"direction\":\"output\",\"data_type\":\"light\",\"id\":\"int_0\"},{\"direction\":\"output\",\"data_type\":\"voltage\",\"id\":\"int_1\"}]}}";
void updateDeviceID(const char* newID) {
    JsonObject& json = jsonBuffer.parseObject(registration);
  //  JsonObject& json = jsonBuffer.createObject();

  json["stored_id"] = newID;
 // Serial.println("json");
 // json.printTo(Serial);
  delay(200);
  char buffer[256];
  json.printTo(buffer, sizeof(buffer));
 // Serial.println();
  delay(200);
 // Serial.println("bufor");
 // Serial.println(buffer);
  registration = String(buffer);
 // Serial.println();
 Serial.println("regiter");
  Serial.println(registration);

  
  
}
void dataSendLight(String dane)
{
  JsonObject& json2 = jsonBuffer.parseObject(SendDataLight);
  json2["value"] = dane;
  Serial.println(SendDataLight);
  Serial.println();
   Serial.println("json");
  delay(100);
  json2.printTo(Serial);
  delay(100);
  char buffer[256];
  json2.printTo(buffer, sizeof(buffer));
  SendDataLight = String(buffer);
 Serial.println();
  delay(100);
  Serial.println(SendDataLight);
   
 //webSocket.sendTXT(SendDataLight);
  
}
void dataSendVoltage(String dane)
{
  JsonObject& json = jsonBuffer.parseObject(SendDataVoltage);
  json["value"] = dane;
 // Serial.println();
  //Serial.println("json");
  delay(100);
 // json.printTo(Serial);
  delay(100);
  char buffer[256];
  json.printTo(buffer, sizeof(buffer));
  SendDataVoltage = String(buffer);
  //Serial.println();
  delay(100);
 // Serial.println(SendDataVoltage);
  // webSocket.sendTXT(SendDataVoltage);

  
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t lenght) {
 
 
 
 
 
 //JsonObject& root = jsonBuffer.parseObject(json);
 //JsonObject& root2 = jsonBuffer.createObject(json);
   /* if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }*/
    switch(type) {
      case WStype_DISCONNECTED:
            USE_SERIAL.printf("[WSc] Disconnected!\n");
            break;
        case WStype_CONNECTED:
            {
            USE_SERIAL.printf("[WSc] Connected to url: %s\n",  payload);  
            //Serial.println(json);      
        webSocket.sendTXT(registration);
        connected = false;
           
            }
            break;
        case WStype_TEXT:
     
    {
            USE_SERIAL.printf("[WSc] get text: %s\n", payload);
            String spayload = (char*)payload;
           JsonObject& json = jsonBuffer.parseObject(spayload); 
            const char* id_device = json ["device_id"];
            Serial.println(id_device);
            updateDeviceID(id_device);
            webSocket.sendTXT(registration);
          
            while(WStype_CONNECTED)
            {
            webSocket.sendTXT(SendDataLight);
            // webSocket.sendTXT(SendDataVoltage);
            
            }
    }
            

      ;
                        
            break;
        
        case WStype_BIN:
        
        
            USE_SERIAL.printf("[WSc] get binary lenght: %u\n", lenght);
            
            hexdump(payload, lenght);
             //char[] dane = analogRead(A0);
            // send data to server
             //webSocket.sendBIN(dane, lenght);
            break;
    
    
}
}

void setup() {
 
  
  configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");
  pinMode(A0,INPUT);
  pinMode(12,OUTPUT);
    // USE_SERIAL.begin(921600);
    USE_SERIAL.begin(115200);
    delay(1000);
 
 
    //Serial.setDebugOutput(true);
    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

      for(uint8_t t = 4; t > 0; t--) {
          USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
          USE_SERIAL.flush();
          delay(1000);
      }

    WiFiMulti.addAP("test", "prezentacja");
    delay(500);

    //WiFi.disconnect();
    while(WiFiMulti.run() != WL_CONNECTED) {
        delay(100);
    }

   //webSocket.begin("iotwebrouter.herokuapp.com", 80, "/device");
    //webSocket.onEvent(webSocketEvent);

}

void loop() {

//webSocket.loop();
  String dataLight;
 dataLight = String(analogRead(A0));

  delay(100);
 Serial.println("data");
 delay(100);
 Serial.println();
 Serial.println(dataLight);
 delay(100);
  dataSendLight(dataLight);
 // dataSendVoltage("0");
 
   
  

  delay(200);

//analogWrite(12, data);

}
```
and i have one more question, how can i replace my edited json with my first json. I use root.printTo(buffer) and conver it to string but it will take a lot of memory wchich i need declare in buffer. Can i print it into json? exp. root.printTo(SendDataVoltage) but its doesnt working.

bblanchon, 2016-01-24:
I'm sorry but I don't understand your questions and I clearly don't have time to read all that code.
Can you please rephrase and isolate the problem?
Thanks.

qblacksheep, 2016-01-24:
I declare a json:
```javascript
String SendDataVoltage =
"{\"action\":\"data\",\"signals\":[{\"interface_id\":\"int_1\",\"message\":{\"data_type\":\"voltage\",\"value\":\"0\"}}]}";
```
I want to change value, need to send it from sensor.
This is my function where i do it:
```javascript
void dataSendLight(String dane)
{
  DynamicJsonBuffer jsonBuffer;
  JsonObject& json2 = jsonBuffer.parseObject(SendDataLight);
  json2["value"] = dane;
  char buffer[256];
  json2.printTo(buffer, sizeof(buffer));
  SendDataLight = String(buffer);
 //webSocket.sendTXT(SendDataLight);
  
}
```
I dont know why but at the output im geting this:
[image]
The value is changing but its paste at the end of my json.
How can i do it propety?
And second question:
I Want overwrite my new json with value from sensor. In this function i print it to buffer, but it will take a lot of memory. I need to print it into the SendDataVoltage in other way but i dont know how.

Im hope u understand my problem now.

bblanchon, 2016-01-24:
Here's your input:

```c++
{
  "action": "data",
  "signals": [
    {
      "interface_id": "int_0",
      "message": {
        "data_type": "light",
        "value": "5"
      }
    }
  ]
}
```

You're updating the wrong field.
Here is the solution to change the value:

```c++
json2["signals"][0]["message"]["value"] = dane;
```

See [FAQ: Parsing succeeds but I can't read the values!](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#parsing-succeeds-but-i-cant-read-the-values).

qblacksheep, 2016-01-24:
Thanks a lot! What about second question? How can i print new value into my input json?'
```javascript
json2.printTo(buffer, sizeof(buffer));
  SendDataLight = String(buffer);
```
Its working but i need a lot of memory to buffer. I tried json2.printTo(SendDataLight) but it didnt working. Have u idea how to print it in other way?

bblanchon, 2016-01-24:
Well... it's not a secret that I hate `String`, so I'd probably write a function like this:

```c++
void sendJson(WebSocket& webSocket, JsonObject& json) {
  char buffer[256];
  size_t size = json.printTo(buffer, sizeof(buffer));
  webSocket.sendTXT(buffer, size);
}
```
But `json2.printTo(SendDataLight)` should have work, what was the problem?

qblacksheep, 2016-01-24:
I'll try to do similar function.
But i have problem with updating value field. I did what u say and wrote it:
```javascript
void dataSendLight(String dane)
{
  JsonArray& json2 = jsonBuffer.parseArray(SendDataLight);
 JsonObject& jsonObj = jsonBuffer.parseObject(SendDataLight);
  // jsonObj["signals"][0]["message"]["value"] = dane;
  json2["signals"][0]["message"]["value"] = dane;
```
and console return me this error:
invalid conversion from 'const char*' to 'size_t {aka unsigned int}' [-fpermissive]

bblanchon, 2016-01-24:
`SendDataLight contains an object, so that's `parseObject()` you want to call.
The commented line is the right one.

See [FAQ: Should I call parseArray() or parseObject()?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#should-i-call-parsearray-or-parseobject)

qblacksheep, 2016-01-24:
Ok, i arleady did it and its working. Thanks. 
I have another problem. I'm sending data by WS and after few min it send just ; "{}". Maybe the buffer is full? I did it by using function which you mentioned: 
```javascript
char buffer[256];
  size_t size = json.printTo(buffer, sizeof(buffer));
  webSocket.sendTXT(buffer, size);
}
```

This is my Serial:
[image]

bblanchon, 2016-01-24:
Please read:

* [Avoiding pitfalls: Don't reuse the same JsonBuffer](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#4-dont-reuse-the-same-jsonbuffer)
* [FAQ: The first parsing succeeds, why does the next ones fail?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#the-first-parsing-succeeds-why-does-the-next-ones-fail)
* [FAQ: I found a memory leak in the library!](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#i-found-a-memory-leak-in-the-library)
* [FAQ: What's the best way to use the library?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#whats-the-best-way-to-use-the-library)

closed 2016-02-01 by bblanchon

suryatmodulus, 2017-10-02:
I know its really late but i hope its useful..... For ESP8266 WebSocketClientIO

#include <ArduinoJson.h> --- use this library

--> payload = 42["arduino",{"message":"forward"}]

case WStype_TEXT:
{
Serial.printf("[WSc] get text: %s\n", payload);
String text = ((const char *)&payload[0]);
int pos = text.indexOf('{');
String json = text.substring(pos,text.length()-1);

        const size_t BUFFER_SIZE =
             JSON_OBJECT_SIZE(10) 
             + MAX_CONTENT_SIZE;              
         DynamicJsonBuffer jsonBuffer(BUFFER_SIZE);

         JsonObject& root = jsonBuffer.parseObject(json);
         String msg = root["message"];

     }

bblanchon, 2017-10-03:
@suryatmodulus thanks for your contribution.

However, this code produces four copies with dynamic memory allocations, which is a waste of CPU time and memory.

Consider using something along those lines:

```c++
// search the begining of the JSON array
char* json = strchr((char*)payload, '[');

// no need to add MAX_CONTENT_SIZE as the content wont be duplicated
const size_t BUFFER_SIZE = JSON_OBJECT_SIZE(10);              
DynamicJsonBuffer jsonBuffer(BUFFER_SIZE);

// since json is a char*, parseObject() wont duplicate the content
JsonObject& root = jsonBuffer.parseObject(json);
const char* msg = root[0]["message"]; // changed to match the example
```

Bonus game: can you spot the four copies in the original program?

muratgortekin, 2017-12-19:
Hello there,
I bought a new server. I want to send and receive multiple data with Arduino. How can I do that? Can you help me with my homework? 💯