bblanchon/ArduinoJson #204: Encoding Problems

schmiddim, 2016-01-22:
At first thanks a lot for your nice library!
I'm encoding JSON data with your from a Http Request and it always failed with UTF-8. 
I could solve the problem in PHP with iconv():

echo iconv("UTF-8", "CP437",trim(json_encode($outputArrays)) );

Could you mention this in your Wiki for other users?

bblanchon, 2016-01-23:
You're right, ArduinoJson has not been designed to support UTF8.
In particular the character escape sequence `\u0000` is not supported.

However, we may find a workaround...
Can you give a specific example?

schmiddim, 2016-01-23:
I'm using an example from your documentation with an ESP8266
 
If the encoding is set to UTF-8 the success method fails:

JsonObject& json_parsed = jsonBuffer.parseObject(json);
      if (!json_parsed.success())
      {
        strip.setPixelColor(NUM_PIXELS - 3, 255, 0, 0);
        strip.show();
        return;
      }

The solution is converting the output to a "more classic" encoding like Codepage 437. 
I solved the Problem with PHP by using the iconv function:
$jsonString = '....';
echo iconv("UTF-8", "CP437", $jsonString) );
I deployed the broken and the running script on theese urls:

broken: http://rock.mercadoultra.com/
works: http://rock.mercadoultra.com/examples/led-matrix.php

The scripts are here on github
https://github.com/schmiddim/mvg-live-php/tree/esp
The concrete Fix is in this file (L47..)
https://github.com/schmiddim/mvg-live-php/blob/esp/examples/led-matrix.php

bblanchon, 2016-01-23:
Here are the results I have when perform request with fiddler:

###### http://rock.mercadoultra.com/

```
HTTP/1.1 200 OK
Date: Sat, 23 Jan 2016 18:42:04 GMT
Server: Apache/2.2.15 (CentOS)
X-Powered-By: PHP/7.0.1
Connection: close
Transfer-Encoding: chunked
Content-Type: text/html;charset=ASCII

e8
{"lines":[{"departures":[5,14,24,34,44,54],"name":"U3","color":{"red":255,"green":127,"blue":0},"departureCount":6},{"departures":[4,14,24,34,44],"name":"28","color":{"red":100,"green":0,"blue":0},"departureCount":5}],"lineCount":2}
0
```

###### http://rock.mercadoultra.com/examples/led-matrix.php

```
HTTP/1.1 500 Internal Server Error
Date: Sat, 23 Jan 2016 18:44:18 GMT
Server: Apache/2.2.15 (CentOS)
X-Powered-By: PHP/7.0.1
Content-Length: 329
Connection: close
Content-Type: text/html; charset=UTF-8

{"lines":[{"departures":[2,12,22,32,42,52],"name":"U3","color":{"red":255,"green":127,"blue":0},"departureCount":6},{"departures":[2,12,22,32,42],"name":"28","color":{"red":100,"green":0,"blue":0},"departureCount":5},{"departures":[3,8,13,18],"name":"27","color":{"red":0,"green":0,"blue":255},"departureCount":4}],"lineCount":3}
```

The problem comes from the server response.

schmiddim, 2016-01-23:
don't care about response code.

bblanchon, 2016-01-24:
Neither do I.

I'm talking about:

1. the charset: ASCII in the first, UTF-8 in the second
2. the `e8` at the beginning of the first message, that looks like a [BOM](https://en.wikipedia.org/wiki/Byte_order_mark)
3. the `0` at the end of the first message

ArduinoJson will reject a JSON input that starts with `e8\n`.

schmiddim, 2016-01-24:
Think it was a mistake on my side - just wanted to write my issue with a solution down that other people can see the solution. 
Best and thanks for your awesome library

bblanchon, 2016-01-24:
You're welcome. Good luck with your project.

closed 2016-01-24 by bblanchon

bblanchon, 2017-02-20:
For the record, the extra line before the JSON was due to [Chunked Transfert Encoding](https://fr.wikipedia.org/wiki/Chunked_transfer_encoding).
It can be turned off by using HTTP 1.0 instead of HTTP 1.1 in the query.