bblanchon/ArduinoJson #205: How to diagnsoe StaticJsonBuffer size limitations?

andig, 2016-01-23:
I'm using `StaticJsonBuffer` as advised and observed some hard- to diagnose issues where by json would be missing this or that parameter. I had of course forgotton to increase the size of my Json buffers as well. 

During development I'd rather see hard exceptions or errors rather than silent omissions. Is it possible to do that?

bblanchon, 2016-01-23:
I'm sorry, but neither exceptions, nor logging would ever be part of ArduinoJson.
Because it would increase the size of the code, which is against the design principles of the library.
Moreover, exceptions are a very controversial feature, especially in the embedded development world.

However, when memory allocation fails, `success()` should return `false`.
If it's not the case, please tell me.

Also remember that you can use `JSON_OBJECT_SIZE()` and `JSON_ARRAY_SIZE()` to compute the required size at compile time.
Of course it means that you need to know the maximum sizes of your objects in advance.
If you cannot predict the size, you must decide a limit yourself: you're working with limited memory, therefore your capacity to decode JSON is limited too.

If you have a lot of memory on your platform, then switch to `DynamicJsonBuffer`.

bblanchon, 2016-01-24:
Here is an implementation of a static decorator adding the feature you're looking for:

```c++
#include <stdexcept>

template <typename TJsonBuffer>
struct Throwing : TJsonBuffer {
  virtual void *alloc(size_t bytes) {
    void *ptr = TJsonBuffer::alloc(bytes);
    if (ptr)
      return ptr;
    else
      throw std::runtime_error("allocation failed");
  }
};

// to be used like this
Throwing<StaticJsonBuffer<200> > jsonBuffer;
// or like this
Throwing<DynamicJsonBuffer> jsonBuffer;
```

I didn't actually tested the code but I'm pretty confident it should work.

bblanchon, 2016-01-24:
PS: Added to the wiki:
https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks#throw-exception-when-jsonbuffer-is-too-small

andig, 2016-01-24:
Excellent, thank you!

closed 2016-02-01 by bblanchon