bblanchon/ArduinoJson #206: Chunked output

sticilface, 2016-01-24:
I'm doing work with async web handlers on the esp8266. 

A desirable feature would be to request a range of the output string. Ie... 

Root.printTo( char * buf, int from, int to) 

This could be done I guess using a similar print class extension as buffered print...but it might be a cool feature to add and would save iterating through the whole json string each time. 

Any interest in adding this overload?

bblanchon, 2016-01-24:
There is two ways to implement this function:

1. store the whole JSON string
2. iterate through the whole json string each time.

The first solution consumes a lot of memory, and the second consumes a lot of CPU cycles.
Since we're working on an embedded platform, we are more constrained in term of memory than CPU cycles.
Therefore, I would choose solution 2.

So, I would have to do exactly the same trick as in the buffered print, except that I would ignore the first chars and only read a certain amount of chars.
There is no better way to do it.

However, it could be a good idea to add these classes to the wiki so that they can be reused.
If there is a lot of interest, maybe I can add them to the source code.

sticilface, 2016-01-24:
Ok. Will do that!

closed 2016-01-24 by sticilface

bblanchon, 2016-01-24:
Here you go my friend:
https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks#chunked-output

sticilface, 2016-01-24:
You beat me to it. Thank you very much:)

sticilface, 2016-01-24:
I'll add the buffered print class to the wiki when I get home.

bblanchon, 2016-01-24:
I did it already.
I kept my original version for Print&, and added a link to the issue you raised.

sticilface, 2016-01-24:
I saw!  thanks:)

sticilface, 2016-01-24:
ok.. so i can't get it to work..

here is my adapted function to write to a buffer instead of print..
```
  class ChunkPrint : public Print
  {
  public:
    ChunkPrint(uint8_t* destination, size_t from, size_t to)
      : _destination(destination), _to_skip(from), _to_write(to - from), _pos{0} {}

    size_t write(uint8_t c)
    {
      if (_to_skip > 0) {
        _to_skip--;
      } else if (_to_write > 0) {
        _to_write--;
        _destination[_pos++] = c;
        return 1;
      }
      return 0;
    }

  private:
    uint8_t* _destination;
    size_t _to_skip;
    size_t _to_write;
    size_t _pos; 
  };
```

I use this to try and write to it.
```
size_t AsyncJsonResponse::_fillBuffer(uint8_t *data, size_t len)
{
  size_t end = _sentLength + len;

  _out->printTo( ChunkPrint(data, _sentLength, end) );

  return len;
}
```

here is the error i get...

```
In file included from Melvanimate-example.ino:65:0:
src/misc.h: In member function 'virtual size_t AsyncJsonResponse::_fillBuffer(uint8_t*, size_t)':
src/misc.h:270:53: error: no matching function for call to 'ArduinoJson::JsonVariant::printTo(ChunkPrint)'
_out->printTo( ChunkPrint(data, _sentLength, end) );
^
src/misc.h:270:53: note: candidates are:
In file included from .pioenvs/nodemcu2/Json_ID64/include/ArduinoJson/Internals/../JsonVariant.hpp:12:0,
from .pioenvs/nodemcu2/Json_ID64/include/ArduinoJson/Internals/../JsonBuffer.hpp:14,
from .pioenvs/nodemcu2/Json_ID64/include/ArduinoJson/Internals/BlockJsonBuffer.hpp:9,
from .pioenvs/nodemcu2/Json_ID64/include/ArduinoJson/DynamicJsonBuffer.hpp:9,
from .pioenvs/nodemcu2/Json_ID64/include/ArduinoJson.h:7,
from .pioenvs/nodemcu2/Json_ID64/ArduinoJson.h:13,
from src/tmp_ino_to.cpp:25:
.pioenvs/nodemcu2/Json_ID64/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:30:10: note: size_t ArduinoJson::Internals::JsonPrintable<T>::printTo(Print&) const [with T = ArduinoJson::JsonVariant; size_t = unsigned int]
size_t printTo(Print &print) const {
^
.pioenvs/nodemcu2/Json_ID64/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:30:10: note:   no known conversion for argument 1 from 'ChunkPrint' to 'Print&'
.pioenvs/nodemcu2/Json_ID64/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:44:10: note: size_t ArduinoJson::Internals::JsonPrintable<T>::printTo(char*, size_t) const [with T = ArduinoJson::JsonVariant; size_t = unsigned int]
size_t printTo(char *buffer, size_t bufferSize) const {
^
.pioenvs/nodemcu2/Json_ID64/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:44:10: note:   candidate expects 2 arguments, 1 provided
.pioenvs/nodemcu2/Json_ID64/include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:49:10: note: size_t ArduinoJson::Internals::JsonPrintable<T>::printTo(String&) const [with T = ArduinoJson::JsonVariant; size_t = unsigned int]
size_t printTo(String &str) const {
```

sticilface, 2016-01-24:
so for whatever reason its not believing my chunked is derived from print and calling 
```size_t printTo(Print &print) const {
```

sticilface, 2016-01-24:
tried a variety of casts and that doesn't work either

```
src/misc.h: In member function 'virtual size_t AsyncJsonResponse::_fillBuffer(uint8_t*, size_t)':
src/misc.h:245:83: error: invalid cast of an rvalue expression of type 'ChunkPrint' to type 'ChunkPrint&'
_out->printTo( (ChunkPrint&)( ChunkPrint(data, _sentLength, _sentLength + len ) ) ) ;
^
```

bblanchon, 2016-01-24:
Try replacing

```c++
_out->printTo( ChunkPrint(data, _sentLength, end) );
```

by 

```c++
ChunkPrint chunkPrint(data, _sentLength, end);
_out->printTo(chunkPrint);
```

That's the best we can do without changing ArduinoJson's source code.

sticilface, 2016-01-24:
thank you...  getting a bit of crashing at the moment.... but small progress

sticilface, 2016-01-24:
so whenever i call printTo I get an exception

```
New Json request
ChunkPrint initialised [0]->[56]Fatal exception 28(LoadProhibitedCause):
epc1=0x40001f1c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007b, depc=0x00000000

Exception (28):
epc1=0x40001f1c epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000007b depc=0x00000000

ctx: sys 
sp: 3ffffa50 end: 3fffffb0 offset: 01a0
```

sticilface, 2016-01-24:
This is the code
```
  ChunkPrint dest( data[0], 0, test.measureLength() ); 
  test.printTo( dest ) ;
```
  class ChunkPrint : public Print
  {
  public:
    ChunkPrint(uint8_t* destination, size_t from, size_t to)
      : _destination(destination), _to_skip(from), _to_write(to - from), _pos{0} {
        os_printf("ChunkPrint initialised [%u]->[%u]", from, to);
      }

     size_t write(uint8_t c)  
    {
      if (_to_skip > 0) {
        _to_skip--;
      } else if (_to_write > 0) {
        _to_write--;
        _destination[_pos++] = c;
        os_printf((const char *)c); 
        return 1;
      }
      return 0;
    }

  private:
    uint8_t* _destination;
    size_t _to_skip;
    size_t _to_write;
    size_t _pos; 
  };
```

sticilface, 2016-01-25:
Got it to work in the end.