bblanchon/ArduinoJson #207: Complex memory requirements

sticilface, 2016-01-25:
me again.... 

I'm working on adapting arduinojson to work with an asynchronous web server on the ESP8266.  I have to overcome two problems.

1) getting chuncked data - which you have provided...
2)  Extending the lifetime of the jsonbuffer, and objects PAST that of the scope of the initial function...  

This is because to send an async web request it is all done via callbacks....  triggered on the ACK of the previous packet... I'm writing a class that deals with the creation of the buffer and the sending of the data, which is fine....  but... my issue is what is the relationship and scope requirements of the jsonobjects themselves and the buffer... 

my class can return a reference to the jsonbuffer.  

but then the user can create an object or an array using `JsonObject& `.. and what I'm not sure about is how far all these objects, and arrays, and sub objects... etc need to be in scope for the integrity of the sending later... or... can you just pass a jsonVarient pointer and use pointer->printTo()? 

Or do all the objects (Object + arrays) need to be in scope for it to work?  
I hope this makes sense? 

having thought i cracked it last night, it is not working now...  Can you correct me if I'm wrong. 
The buffer has to stay in scope for the object to be able to print.  BUT...  the reference to the original object can go out of scope... as long as you've stored it somewhere, say as a pointer it will work...  Through a lot of trial and error this seems to work... 

However, I would like this to be flexible... so i've tried to write a function like before
```
template <class T> void AsyncJsonResponse::SetTarget( T & root)
{
  _root = &root;
  _contentLength = _root->measureLength();

}
```

where _root is a pointer to JsonObject... now this works... but if i change the pointer to JsonVarient (so it can store a pointer to object and Array) i get garbage... 

What is the best way for this class to take a Json item.. (object or Array)  store it so that it can be used for the measurelength() and printTo()..

Let me know if this makes sense?

bblanchon, 2016-01-25:
I'm missing the big picture...
Can you show me how all this code has to be invoked?
Can you copy a snippet of asynchronous web server?

sticilface, 2016-01-25:
of course... 

This class is derived from a generic set of response classes...  which is called to send the data, keep track of how much has been sent etc... and fill the buffer with the next chunk.. 

```
class AsyncJsonResponse: public AsyncAbstractResponse
{
private:

  DynamicJsonBuffer _jsonBuffer;
  //JsonObject ** _out;
  JsonObject* _root;

  class ChunkPrint : public Print
  {
  public:
    ChunkPrint(uint8_t* destination, size_t from, size_t to)
      : _destination(destination), _to_skip(from), _to_write(to - from), _pos{0}
    {
      //os_printf("[ChunkPrint] ChunkPrint initialised [%u]->[%u]\n", from, to);
    }

    size_t write(uint8_t c)
    {
      if (_to_skip > 0) {
        _to_skip--;
      } else if (_to_write > 0) {
        _to_write--;
        _destination[_pos++] = c;
        return 1;
      }
      return 0;
    }

  private:
    uint8_t* _destination;
    size_t _to_skip;
    size_t _to_write;
    size_t _pos;
  };

public:

  AsyncJsonResponse();
  ~AsyncJsonResponse();

  DynamicJsonBuffer & getBuffer() { return _jsonBuffer; }
  bool _sourceValid();
  template <class T> void SetTarget(  T & root);

  size_t _fillBuffer(uint8_t *buf, size_t len);

};

bool AsyncJsonResponse::_sourceValid()
{
  if (_root) {
    return true; 
  }

  return false;
}

AsyncJsonResponse::~AsyncJsonResponse()
{
  //os_printf("[~AsyncJsonResponse]\n");
}

template <class T> void AsyncJsonResponse::SetTarget( T & root)
{
  _root = &root;
  _contentLength = _root->measureLength();

}

AsyncJsonResponse::AsyncJsonResponse() : _root{nullptr}
{
  _code = 200;
  _contentType = "text/json";
}

size_t AsyncJsonResponse::_fillBuffer(uint8_t *data, size_t len)
{
//  os_printf("[_fillBuffer] len = %u, heap = %u\n", len, ESP.getFreeHeap() );
  ChunkPrint dest(data, _sentLength, _sentLength + len );
//  os_printf("[_fillBuffer] chunk created, start [%u] -> end [%u]\n", _sentLength, _sentLength + len);
  _root->printTo( dest ) ;
  return len;

}
```

so _fillbuffer is called every ACK..  until the _contentLength is reached. 

here is the setup for the web server... 

```
  HTTP.on("/json", HTTP_ANY, [](AsyncWebServerRequest * request) {

    Serial.println();
    Serial.printf("New Json request: heap = %u\n", ESP.getFreeHeap()); 

   // Create new async response... 
    AsyncJsonResponse * response = new AsyncJsonResponse(); 
// get new buffer from object..
    DynamicJsonBuffer buffer = response->getBuffer(); 
// create expendable objects, by reference... 
    JsonObject& root = buffer.createObject(); 
    JsonObject& sub1 = root.createNestedObject("nested1");
    JsonObject& test = sub1.createNestedObject("nested2");
// add some data...
   test["2"] = "String1";
   test["3"] = "String1";
   test["4"] = "String1";

//  set the target you want to print.. this assigns pointer, and calculates content length... 
    responce->SetTarget(root);

// set up call back,,,
     request->send(responce); 
  });
```

sticilface, 2016-01-25:
So this works.. i just want to be able to store / print any json object by SetContent()

i guess templates are the way forward..

bblanchon, 2016-01-25:
You can change `_root` to be a `JsonVariant`, but don't use a pointer.

```c++
class AsyncJsonResponse: public AsyncAbstractResponse
{
private:
   JsonVariant _root;

public:
   void SetRoot(JsonVariant root) {
      _root = root;
   } 
};
```

sticilface, 2016-01-25:
ah.. man... always the simplest solutions... Thank you

bblanchon, 2016-01-25:
This is pretty good stuff that you have here.
It would definitely find a place in the [Bag of Tricks](https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks)...

However, I would personally not nest `ChunkPrint` into `AsyncJsonResponse` as it makes sense in its own.
Also, I would compose `ChunkPrint` and `ArduinoJson::Internals::StaticStringBuilder` to implement `_fillBuffer()`:

```c++
size_t AsyncJsonResponse::_fillBuffer(uint8_t *data, size_t len)
{
  StaticStringBuilder stringBuilder(data, len);
  ChunkPrint dest(stringBuilder, _sentLength, _sentLength + len);
  return _root.printTo(dest);
}
```

That way the application is build of composable blocks of reusable code.

sticilface, 2016-01-25:
cool cheers:)

This is totally experimental stuff... its thanks to @me-no-dev who has put all the work in by creating the async-webserver... which is looking pretty awesome...  but totally not mainstream yet!  we did  implement a stream class for the asyncserver... but it was basically a hack that turned the response into a sync response...  obviously blocking the ESP.....  not ideal....  

I then had an eureka moment, whilst away.. and realised i was tackling it from the wrong angle!!! 

I'm not sure i follow the need for StaticStringBuilder.. except that i;ve not shoehorned it in... what does that offer, over just copying in the data?

bblanchon, 2016-01-25:
The idea behind `StaticStringBuilder` (which is a class that already exists in ArduinoJson) is to keep `ChunkPrint` generic by working on a `Print&` (and not on a `uint8_t`, which is more restrictive)

sticilface, 2016-01-25:
ah ok...  hence it would work with your original chunked print?

bblanchon, 2016-01-25:
Yes, at the cost of a virtual call.
But we could easily avoid this cost by using templates.

sticilface, 2016-01-25:
can't seem to get it to work nicely though... not getting any response back.. 

```
class AsyncJsonResponse: public AsyncAbstractResponse
{
private:
  DynamicJsonBuffer _jsonBuffer;
  JsonVariant _root;
  bool _isValid;

public:

  AsyncJsonResponse() : _isValid{false}
  {
    _code = 200;
    _contentType = "text/json";
  }
  ~AsyncJsonResponse() {};

  DynamicJsonBuffer & getBuffer() { return _jsonBuffer; }
  bool _sourceValid() { return _isValid; }
  void SetTarget( JsonVariant root)
  {
    _root = root;
    _contentLength = _root.measureLength();
    if (_contentLength) { _isValid = true; }
  }
  size_t _fillBuffer(uint8_t *data, size_t len)
  {
    //ChunkPrint dest(data, _sentLength, _sentLength + len );
    //_root.printTo( dest ) ;
    //return len;

    StaticStringBuilder stringBuilder( (char*)data, len);
    ChunkPrint dest(stringBuilder, _sentLength, _sentLength + len);
    return _root.printTo(dest);

  }

};
```

bblanchon, 2016-01-25:
It looks OK to me.

sticilface, 2016-01-25:
yeah... but i get nada

sticilface, 2016-01-25:
ok, so its getting stuck on the last character...
```
{"nested1":{"nested2":{},"key2":"abcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefge"},"array":["This is some randome text to fill the array","This is some randome text to fill the array","This is some randome text to fill the array","This is some randome text to fill the array","This is some randome text to fill the array"],"key1":"abcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefgeabcdefge"}pm open,type:2 0
}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}}
```

i put `      os_printf("%c",c); ` into the chunked output .  So this is the server realising it is one letter short and it continues to request it

sticilface, 2016-01-25:
does your string builder expect a null end.. so it stops one short?

sticilface, 2016-01-25:
ok fixed it... 
```
    StaticStringBuilder stringBuilder( (char*)data, len+1);
```

sticilface, 2016-01-25:
I had to change you print instances in the chunkedoutput to references... otherwise i got pure virtual warnings... might be work changing in the wiki too?

bblanchon, 2016-01-25:
Suer go ahead, but were is it? All I see is reference.

sticilface, 2016-01-25:
```
class ChunkPrint : public Print {
 public:
  --->ChunkPrint(Print& destination, size_t from, size_t to)
      : _destination(destination), _to_skip(from), _to_write(to - from) {}

  virtual size_t write(uint8_t c) {
    if (_to_skip > 0) {
      _to_skip--;
    } else if (_to_write > 0) {
      _to_write--;
      return _destination.write(c);
    }
    return 0;
  }

 private:
--->  Print& _destination;
  size_t _to_skip;
  size_t _to_write;
};
```
just where i've put --->  i had to change it to print&

bblanchon, 2016-01-25:
I don't understand, [these ampersands have been there from the beginning](https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks/_history).
Could this just be a display issue?

sticilface, 2016-01-25:
You are right!  no idea...  i swore i copied it directly from you page... when you suggested to use print& instead.. i went back to the wiki and copied the exact function...  never mind... it is working now :)

sticilface, 2016-01-26:
Seem to have another issue, where a large son causes a crash.  If i just use printTo(Serial) the same string does not crash.  However, the crashing function is within the arduinojson internals... see
https://github.com/bblanchon/ArduinoJson/issues/208

bblanchon, 2016-01-26:
I wonder if this could be the zero terminator added by `StaticStringWriter`...
Does the crash appears with your version of `ChunkPrint` that works directly with `uint8_t*`?

sticilface, 2016-01-26:
That is what I thought. But it happens with both!

bblanchon, 2016-01-26:
Can you share the complete source code (a [Gist](https://gist.github.com/) maybe), so I can walk through it?

sticilface, 2016-01-26:
Can't right now, at work. But later. 
Are you able to use the arduino IDE, and esp8266 build environment? 

I will make a simple sketch, too, mine is quite complex!

sticilface, 2016-01-26:
Can you tell which function is causing it. I had a look but couldn't find it

bblanchon, 2016-01-26:
At home, I have a [Sparkfun Thing](https://www.sparkfun.com/products/13231) ready, would that be OK?
Even if I can't build the code, reviewing the code may help us find the issue.

sticilface, 2016-01-26:
That will do nicely! You'll have to get some dependencies. Arduinojson for example:) plus the async libs.

sticilface, 2016-01-26:
So here is a gist.  Let me know how you get on.

https://gist.github.com/sticilface/ceef9a0b031afc0ce2a2

There is a function addJson... if you uncomment the last few lines that is enough for the chunkedprint/async sending to crash, whereas printing to serial does not.

bblanchon, 2016-01-26:
There is nothing obvious at a first sight.
I'll test when I'll be home.

bblanchon, 2016-01-26:
Did you try to send such a large string without using ArduinoJson, maybe it's a problem with the async lib?

sticilface, 2016-01-26:
i realise that this is the 'first version' with nested chunkedprint... but i reverted when this error threw its head!

sticilface, 2016-01-26:
i don't think it is the string size... as the async server only requests chunks...i think 2096bytes or so...

sticilface, 2016-01-26:
are you on gitter?

bblanchon, 2016-01-26:
I am now, and I created the chat room

bblanchon, 2016-01-26:
So, as I understood, the crashes are solved now, there were due to an issue in the async library.
Now remains a speed issue that we'll try to address with #210.

closed 2016-01-26 by bblanchon

sticilface, 2016-01-26:
correct!  
(and it was my coding blunder that caused the crashes.. a memory leak, which corrupted the buffer!

bblanchon, 2016-01-26:
Still there @sticilface? do you even sleep?

sticilface, 2016-01-26:
I am still here! Of course I sleep:)