bblanchon/ArduinoJson #212: How I can reuse JsonObject variables?

4refr0nt, 2016-01-28:
Hi!
How I can set, then empty and reuse JsonObject variables

```
StaticJsonBuffer<1024> jsonBuffer;
JsonObject& root = jsonBuffer.createObject();
root["foo"] = "foo";
```
result {"foo":"foo"}

```
......
root = jsonBuffer.createObject(); // empty???
root["bar"] = "bar";
```
I need result {"bar":"bar"}, not {"foo":"foo", "bar":"bar"}

bblanchon, 2016-01-29:
Hi Victor,

This is a common misconception that it's more efficient to reuse objects. but in the case of ArduinoJson, it's not.

The best way to have an empty `JsonObject` is to create a new one.
It's even better if you start from a fresh `JsonBuffer`, so that you can be sure that all the memory has been freed.
Also, reference in C++ cannot be reaffected, so you need to declare a new reference.

Example:

```c++
JsonObject& obj1 = jsonBuffer.createObject();
obj1["foo"] = "foo";

JsonObject& obj2 = jsonBuffer.createObject();
obj1["bar"] = "bar";
```

closed 2016-01-29 by 4refr0nt

domdetre, 2018-08-06:
@4refr0nt I know, years passed, but it is mainly for other people seeking for a REAL answer:
`
JsonObject* root = &jsonBuffer.parseObject(jsonString);
root = &root->createNestedObject("somekey");
String theValue = (*root)[key];
`
also
`
JsonObject& child = (*root)["somekey"];
root = &child;
`

because sometimes you need to dynamically (and not hardcoded-ly) go to a child object in a for loop where you need to reassign.