bblanchon/ArduinoJson #214: `is<T>()` returns wrong value, only with bool

marvinroger, 2016-01-31:
I am using an ESP8266 (NodeMCU 1.0) with the following sketch:

```c++
#include <ArduinoJson.h>

const char* json = "{\"ota\": {\"enabled\": true}}";

void setup() {
  Serial.begin(115200);
  Serial.println();

  StaticJsonBuffer<JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(1)> jsonBuffer;
  JsonObject& parsedJson = jsonBuffer.parseObject(json);
  if (parsedJson["ota"]["enabled"].is<bool>()) {
    Serial.println("ota.enabled is a bool");
  } else {
    Serial.println("ota.enabled is not a bool");
  }
}

void loop() {
  delay(1000);
}
```

This prints `ota.enabled is not a bool`, but it obviously is.
Note I tested with `const char*`, `unsigned long`, `JsonObject&` and it works well, the problem only appears with `bool`.

bblanchon, 2016-01-31:
The bug is real, but the code sample is wrong.
The `StaticJsonBuffer` is too small to hold a copy of the input.
Remember: `JsonBuffer` will have to make a copy if the input is not mutable, ie if it's a `const char*` or a `String`.

Here is the fixed code:

```c++
#include <ArduinoJson.h>

char json[] = "{\"ota\": {\"enabled\": true}}";

void setup() {
  Serial.begin(115200);
  Serial.println();

  StaticJsonBuffer<JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(1)> jsonBuffer;
  JsonObject& parsedJson = jsonBuffer.parseObject(json);
  if (parsedJson["ota"]["enabled"].is<bool>()) {
    Serial.println("ota.enabled is a bool");
  } else {
    Serial.println("ota.enabled is not a bool");
  }
}

void loop() {
  delay(1000);
}
```

marvinroger, 2016-01-31:
Right, my C++ background is a bit limited! But in my code I'm actually using mutable strings.

bblanchon, 2016-01-31:
I fixed the bug, I just need to merge and make a new release

marvinroger, 2016-01-31:
Great, thanks for the quick fix.

bblanchon, 2016-01-31:
Fix in version 5.0.8

closed 2016-01-31 by bblanchon