bblanchon/ArduinoJson #216: Limited of JsonArray is 10?

mrhieuit, 2016-02-01:
I try parse a JsonArray from a string. It Ok. Ex:

char* text = "{"id":"01", "action": [1111,2222,33333,4444,5555,6666,7777,8888,9999,1000]}";
JsonObject& item = jsonBuffer.parseObject(text );
JsonArray& action = item["action"].asArray();
Serial.println(action.size());

However, I get a problem when increse size of JsonArray to larger 10 item Ex:
char* text = "{"id":"00012", "action": [1111,2222,33333,4444,5555,6666,7777,8888,9999,1000, 11,12,13,14]}";
JsonObject& item = jsonBuffer.parseObject(text );
JsonArray& action = item["action"].asArray();
Serial.println(action.size());
--> I get exception.

Limited of JsonArray is 10? Please help me.

bblanchon, 2016-02-01:
No, the size of `JsonArray` is not limited.

Please have a look at [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails).

I bet that you're using the `StaticJsonBuffer<200>` from the example, you probably just need to increase that value.

mrhieuit, 2016-02-01:
Yes, My problem is StaticJsonBuffer<200>. My problem is solved when my increase that value.
Thanks so much. Your lib is great!

bblanchon, 2016-02-01:
You're welcome.
I [added comments in the examples](https://github.com/bblanchon/ArduinoJson/commit/ce63e9c3c3c58d908d12c5728057a1b4c224eba8), so it should be clearer now.

closed 2016-02-01 by bblanchon