bblanchon/ArduinoJson #217: How to convert JsonArray to Int Array

mrhieuit, 2016-02-01:
I have a Json value, I can parse and get JsonArray nested:

char* value =  "{\"id\":\"01\", \"data\": [1,2,3,4,5,6,7,8,9,10]}\";
JsonObject& item = jsonBuffer.parseObject(value);
JsonArray& data= item["data"].asArray();

Now I want convert data from JsonArray to int[ ]. How I do it?
Thanks so much.

bblanchon, 2016-02-01:
Have a look at [Example 3: Iterators](https://github.com/bblanchon/ArduinoJson/wiki/Examples#example-3-iterators)

mrhieuit, 2016-02-02:
Yes, I can do it with for loop and cast for each item to int. However, I think ArduinoJson Library should have method to fast covert like: 

int[] intArray = root["data"].asArray<int>();

I infer idea from your examable: long time = root["time"].as<long>();

bblanchon, 2016-02-02:
The line you wrote is clearly Java or C#:

```java
int[] intArray = root["data"].asArray();
```

The syntax in C++ would be:

```java
int intArray[42] = root["data"];
```

Thus requiring to know the size in advance

To have a dynamic size, we'd have to write something link that:

```c++
std::vector<int> intArray = root["data"];
```

But this would introduce a dependency on `std::vector`, which is not available on Arduino.

And anyway, it would not be a correct C++ idiom. Instead, the C++ way of doing this is:

```c++
std::vector<int> intArray;
JsonArray& jsonArray = root["data"];
std::copy(jsonArray.begin(), jsonArray.end(), std::back_inserter(intArray));
```

Which is probably already possible with current implementation of ArduinoJson.

mrhieuit, 2016-02-04:
I got it. Thanks for the helpful answers.

marvinroger, 2016-02-06:
@bblanchon I guess you can close this one.

closed 2016-02-07 by bblanchon