bblanchon/ArduinoJson #218: How to extract only the repsonse code?

Mat2Vence, 2016-02-02:
Hi,
I am wondering if there is a way to extract only the response code of a received JSON response?
Thanks for your help.

bblanchon, 2016-02-03:
Hi,
I'm sorry I don't undersand your question...
Can you elaborate?

Mat2Vence, 2016-02-03:
Hi,
Sorry if I was not clear enough. 
In fact in the HTTP header of the JSON request, there is a status code that is passed depending on the situation. This is a 200 status if the request is OK, but for instance, this is a status 204 if the request does not send back any data.
My question is related to this status. This is more convenient to parse first only the header of the response in order to identify what is the response status. Then depending on it (rejected, OK, no data...) the rest of the JSON could be parsed or not.
Hope I have been a bit more clear :)

bblanchon, 2016-02-03:
Well HTTP and JSON are different things.
ArduinoJson solves the JSON part.
For the HTTP part, you'll have to deal with it by yourself or find another library.
Did you try with `scanf`?

marvinroger, 2016-02-06:
Are you using an ESP8266?

JoeAlamo, 2016-02-06:
HTTP responses are formatted in a specific way where the response code will be at a certain part (i.e. after x spaces), loop through response character by character counting the spaces and then you'll know when you've hit the response code

closed 2016-02-10 by bblanchon