bblanchon/ArduinoJson #220: Storing JSON to string or Char

shirish47, 2016-02-08:
I want to use this lib with ESP8266HTTPclient 
with the function
`String pl="lets try this";
int httpCode =http.sendRequest("POST",(uint8_t*)pl.c_str(),pl.length());`

where pl must be string converted from JSON.

is there a fuction in this lib? that  I can use directly with this function. don't want to do `json.printTo(client);`

Thanks. :)

bblanchon, 2016-02-08:
Assuming that you have the following object:

```c++
JsonObject& root;
```

Then, you can do:

```c++
String pl;
root.printTo(pl);
```

Or:

```c++
char pl[256];
size_t n = root.printTo(pl, sizeof(pl));
```

The last one being the most efficient in term of code size and speed.

@shirish47, did you look at the documentation before asking this question?

shirish47, 2016-02-08:
thank you I tried the first solution and it works nicely.

Thanks.

closed 2016-02-08 by shirish47

bblanchon, 2016-02-08:
@shirish47, did you look at the documentation before asking this question?

shirish47, 2016-02-10:
yes I am sorry I tried to find out about it... it(ESP8266) did not work for first time .. and got stuck . I then tried it again and it worked.