bblanchon/ArduinoJson #221: How to read this json project with multiple sensors?

wsun0928, 2016-02-10:
I am able to follow this json project. I am trying to put a couple of sensors on one arduino, which makes my json look like {"device":"temp", "value":"22"},{"device":"lux","value":"888"} something like that. I am trying to make arduino read only one {}. Can anyone help me? thank you

bblanchon, 2016-02-10:
Reading only the first one works without any trick.

See https://github.com/bblanchon/ArduinoJson/blob/issue221/test/Issue221.cpp:

```c++
char json[] =
  "{\"device\":\"temp\", "
  "\"value\":\"22\"},{\"device\":\"lux\",\"value\":\"888\"}";
DynamicJsonBuffer jsonBuffer;
JsonObject& obj1 = jsonBuffer.parseObject(json);
ASSERT_TRUE(obj1.success());
ASSERT_STREQ(obj1["device"], "temp");
ASSERT_EQ(obj1["value"], 22);
```

However, I have no idea how to read the second one.

It would have been easier if both objects were contained in an array, or if the separator was not a character that is already present in the JSON object.

wsun0928, 2016-02-10:
I was thinking of 2 methods. 
1. only read one bracket at a time, like only read from x bit to y bit...not sure if it is possible 
2. put them in to arrays, ex: {device:[temp,lux],values:[25,888]} ...the issue is that how can the code identify them as "pair"..
That json is created by the table I made in php. Not sure how to process them separately. 
[{"sensor":"Thermometer","value":"87"},{"sensor":"Humidity","value":"55"},{"sensor":"Motion Room 1","value":"1"},{"sensor":"Lux","value":"7000"}]

bblanchon, 2016-02-10:
Why don't you call `parseArray()`?

FrankOcean11, 2016-02-10:
im going to be in the same boat soon as im looking to read values from a .json file which contains values that i want to be read and input into a LED parameter to control dim according to the array of this .json file which may consist of 11 diff values and of course my goal is to read .json file and write that read value to the write.led(.json); parameter with a delay(1500); ms .json in the led write parameter will be a array named LED[11];  

bblancohon im not sure if parseArray() will work for me OR if my described approach is logical ?

bblanchon, 2016-02-12:
@FrankOcean11 `parseArray()` will work as long as the JSON input file contains a valid JSON array.

For instance, if the input is like this:

```json
[255,127,230]
```

or

```json
[{"port":1,"dim":255},{"port":2,"dim":127},{"port":3,"dim":230}]
```

Then `parseArray()` will work.

FrankOcean11, 2016-02-13:
do you know if theirs a program that converts or reformat your doc /txt or any file into json ? thanks for the input though

bblanchon, 2016-02-18:
@wsun0928 any progress? should I close the issue?
@FrankOcean11 no, i don't think such a thing exists

wsun0928, 2016-02-18:
@bblanchon  I will try the parseArray() this weekend. I am caught up with exams this week. Just to be clear, you are saying that if I use only numbers instead of strings, it will be able to read the brackets individually?

bblanchon, 2016-02-18:
@wsun0928 I meant that life would be easier if you surround your results by square brackets:

```json
[{"device":"temp", "value":"22"},{"device":"lux","value":"888"}]
```

because you could treat the whole as an array of objects.

hemantsangwan, 2016-02-19:
@wsun0928 : I support https://github.com/bblanchon/ArduinoJson/issues/221#issuecomment-185915464 . It will keep system straightforward and simple.

closed 2016-03-02 by bblanchon