bblanchon/ArduinoJson #222: Error using BLE Nano board via arduino

Vineetsai, 2016-02-10:
Running one of example for BLE Nano board, get compilation error.

> In file included from /Users/vineet/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:10:0,
                 from /Users/vineet/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from /Users/vineet/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:8,
                 from /Users/vineet/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from /var/folders/ty/xh2f71fs651cwpnrj4tykm05_04_1g/T/arduino_2544ff0d6a27032e23f1a3b6df22e3e3/JsonGeneratorExample.ino:8:
/Users/vineet/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function 'ArduinoJson::JsonArray& ArduinoJson::JsonBuffer::parseArray(const String&, uint8_t)':
/Users/vineet/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:71:28: error: 'const class String' has no member named 'c_str'
     return parseArray(json.c_str(), nesting);
                            ^
/Users/vineet/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function 'ArduinoJson::JsonObject& ArduinoJson::JsonBuffer::parseObject(const String&, uint8_t)':
/Users/vineet/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:94:29: error: 'const class String' has no member named 'c_str'
     return parseObject(json.c_str(), nesting);
                             ^
/Users/vineet/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function 'char* ArduinoJson::JsonBuffer::strdup(const String&)':
/Users/vineet/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:102:25: error: 'const class String' has no member named 'c_str'
       return strdup(src.c_str(), src.length()); 
                         ^
exit status 1
Error compiling.

Vineetsai, 2016-02-10:
found its similar to https://github.com/bblanchon/ArduinoJson/issues/123
Should be fine to mark duplicate.

bblanchon, 2016-02-11:
Apparently the source code of the Arduino Core for your BLE Nano is located here:
https://github.com/RedBearLab/nRF51822-Arduino

And more precisely, the `String` class is declared here:
https://github.com/RedBearLab/nRF51822-Arduino/blob/S130/arduino-1.6.x/hardware/RBL/RBL_nRF51822/cores/RBL_nRF51822/WString.h

Here are the differences between the original `String` from Arduino and this one from RedBearLab:
https://www.diffchecker.com/ccyy2vh4

As you can see `c_str()` is missing, and this is what is causing the compilation error.

bblanchon, 2016-02-11:
I created a [pull request](https://github.com/RedBearLab/nRF51822-Arduino/pull/57).
Meanwhille, you should add the missing function manual on your local copy.

bblanchon, 2016-02-11:
Added to the wiki:
https://github.com/bblanchon/ArduinoJson/wiki/Compatibility-issues#redbearlab-boards-compilation-error

Vineetsai, 2016-02-11:
Yes, that what I did for the time being to make it work.

bblanchon, 2016-03-02:
20 days, and still no reply from RedBearLab :astonished:

I'm closing this issue because there is nothing else I can do.

closed 2016-03-02 by bblanchon

bblanchon, 2016-03-03:
The PR has been merged :-)