bblanchon/ArduinoJson #224: How to read this json struct

tablet23, 2016-02-15:
Hello,
I need to parse the next JSON:
"{"data":[{"value":"18"}, {"value":"20"}, {"value":"50"},....{"value n":"n"}]}";

The size of Json is variable.
How can I do this?

Regards.

bblanchon, 2016-02-15:
Hello,

```c++
char json[] = "{\"data\":[{\"value\":\"18\"}, {\"value\":\"20\"}, {\"value\":\"50\"}]}";
DynamicJsonBuffer jsonBuffer;
JsonObject& root = jsonBuffer.parseObject(json);
JsonArray& data = root["data"];
for (JsonObject& item : data) {
  Serial.println(item["value"].as<int>());
}
```

Did you look in the documentation before asking?

Regard.

TonvanWissen, 2016-02-16:
Hi Peter,
thanks for your very well documented library.
Although I didn't find an answer for this: "How do I decode an array of Json objects.?"

My Json array has 9 rows and looks like this:
[{"id":"A","X_On":"0","X_En":"1","X1i":"hh:mm","X1o":"hh:mm","X2i":"hh:mm","X2o":"hh:mm"},{"id":"B","X_On":"0","X_En":"1","X1i":"hh:mm","X1o":"hh:mm","X2i":"hh:mm","X2o":"hh:mm"},{"id":"C","X_On":"0","X_En":"1","X1i":"hh:mm","X1o":"hh:mm","X2i":"hh:mm","X2o":"hh:mm"},{"id":"D","X_On":"0","X_En":"0","X1i":"hh:mm","X1o":"hh:mm","X2i":"hh:mm","X2o":"hh:mm"},{"id":"E","X_On":"0","X_En":"0","X1i":"hh:mm","X1o":"hh:mm","X2i":"hh:mm","X2o":"hh:mm"},{"id":"F","X_On":"0","X_En":"0","X1i":"hh:mm","X1o":"hh:mm","X2i":"hh:mm","X2o":"hh:mm"},{"id":"G","X_On":"0","X_En":"0","X1i":"hh:mm","X1o":"hh:mm","X2i":"hh:mm","X2o":"hh:mm"},{"id":"H","X_On":"0","X_En":"0","X1i":"hh:mm","X1o":"hh:mm","X2i":"hh:mm","X2o":"hh:mm"},{"id":"I","X_On":"0","X_En":"0","X1i":"hh:mm","X1o":"hh:mm","X2i":"hh:mm","X2o":"hh:mm"}]

How to do this?

Thank you

Kind regards, Ton

bblanchon, 2016-02-16:
Hi @TonvanWissen ,

In your case, you need to call  `parseArray()` instead of `parseObject()`.

```c++
DynamicJsonBuffer jsonBuffer;
JsonArray& root = jsonBuffer.parseArray(json);
for (JsonObject& item : data) {
  const char* id = item["id"];
  const char* X_On = item["X_On"]
  // ...
}
```

Don't hesitate to edit the wiki, the example section in particular.

Regards,
Benoit

closed 2016-02-19 by bblanchon