bblanchon/ArduinoJson #225: support for long long gives compiler error

cyberreefguru, 2016-02-16:
Hi Benoit - per your email, here is a new ticket about long long support for my platform.  Hopefully I can provide enough information:

OS: Mac OS, 10.10.5
Platform: ESP8266
Compliler: esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++
Combiner: tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc

Compiler does not appear to find long long as a template pattern:

```
ArduinoJson/include/ArduinoJson/Internals/../JsonVariant.hpp:153:3: error: no type named 'type' in 'struct ArduinoJson::TypeTraits::EnableIf'
In file included from /ArduinoJson/include/ArduinoJson/JsonObject.hpp:153:0,
from /ArduinoJson/include/ArduinoJson/JsonArray.ipp:11,
from /ArduinoJson/include/ArduinoJson/JsonArray.hpp:181,
from /ArduinoJson/include/ArduinoJson.h:9,
from /ArduinoJson/ArduinoJson.h:14,
from ../Command.h:15,
from ../Command.cpp:8:
/ArduinoJson/include/ArduinoJson/JsonObject.ipp: In member function 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = long long int]':
/ArduinoJson/include/ArduinoJson/JsonObject.ipp:25:1: warning: control reaches end of non-void function [-Wreturn-type]

```
I will attempt to create a small, stand alone version of the app that reproduces the problem.

-Tom

bblanchon, 2016-02-17:
I just realized that this it exactly the kind of error you get when you try to use `long long` when `ARDUINOJSON_USE_LONG_LONG` is set to `0`.

I didn't understand that you were using `long long` in your program.
Be careful, it will require a lot of memory.

If I'm right, the fix is to set `ARDUINOJSON_USE_LONG_LONG` to `1`, preferably as a flag on the compiler command line, to be sure that all .cpp files are compiled with the same configuration.

Also, you should switch to `DynamicJsonBuffer` because the stack size is quite limited on the ESP8266

bblanchon, 2016-02-26:
@cyberreefguru Can I close this issue?

cyberreefguru, 2016-02-28:
Yes.  Thanks!

bblanchon, 2016-03-02:
You're welcome, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add you projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).
Also, a GitHub star is always appreciated :wink:

closed 2016-03-02 by bblanchon