bblanchon/ArduinoJson #226: [parseArray] How to access to key and value on array of object

dade80vr, 2016-02-17:
Hi. Please look at this Json parsed with parseArray():

`[
  {
    "room1": {
      "date": "2016-01-31 17:30:28",
      "temp": "17.5"
    }
  },
  {
    "room2": {
      "date": "2016-02-16 00:00:00",
      "temp": "19"
    }
  }
]
`

How can i access to key name (for example "room1") and sub-object "temp" value (17.5) ??
If i use root[0] or root{0} it does not work!
Thanks and best regards.

bblanchon, 2016-02-18:
Hi Davide,

The only way to get the name of the keys it through iterators:

```c++
const char* name1 = root[0].begin()->key; // "room1"
const char* name2 = root[1].begin()->key; // "room2"
```

...which is horrible :disappointed:

The pain we're filling here is due to the strange JSON layout.
I would suggest to change it to something like:

```json
[
  {
    "location": "room1",
    "date": "2016-01-31 17:30:28",
    "temp": "17.5"
  },
  {
    "location": "room2",
    "date": "2016-02-16 00:00:00",
    "temp": "19"
  }
]
```

That would make more sense.
It would be easier to deal with:

```c++
const char* name1 = root[0]["location"]; // "room1"
const char* name2 = root[1]["location"]; // "room2"
```

And that is true for any JSON library, not only ArduinoJson.

Regards.

dade80vr, 2016-02-21:
Yes, it's working.
Thanks for the tip.

closed 2016-02-21 by dade80vr