bblanchon/ArduinoJson #227: ESP12 Exception (29)

alinleonard, 2016-02-18:
Hi I got this error when I try to power esp.

```
Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff09e0 end: 3fff0e10 offset: 01a0

>>>stack>>>
3fff0b80:  3fff2050 00000182 40233fb0 4020bc0a  
3fff0b90:  40212ced 3fff2010 00000000 ffffff80  
3fff0ba0:  65736724 4020e1db 3fff2010 3fff1a90  
3fff0bb0:  40212f2e 3fff2010 3fff1a90 00000000  
3fff0bc0:  3ffefddc 3ffed6e0 3ffed65c 3fff1a90  
3fff0bd0:  00000000 00000000 00000001 00000000  
3fff0be0:  00000018 00000064 da34fe1a fffeffff  
3fff0bf0:  3ff20a00 0000ffff 3fff0c18 4020e882  
3fff0c00:  3ffed65c 3fff1a90 3fff1a90 00000000  
3fff0c10:  da34fe1a 3fff127f 00000000 00000000  
3fff0c20:  00000000 00000000 00000000 00000000  
3fff0c30:  00000000 00000000 00000000 00000000  
3fff0c40:  00000000 00000000 00000000 4020f9d4  
3fff0c50:  3fff1a90 00000001 00000001 00000000  
3fff0c60:  00000000 3ffed6e0 3ffeeaf8 00000000  
3fff0c70:  402277ad 00000003 00000003 0000007d  
3fff0c80:  402278f9 00000003 00000001 00000000  
3fff0c90:  3fff1240 4022796e 00000003 00000001  
3fff0ca0:  402025dd 3ffefba0 00000001 00000000  
3fff0cb0:  40202614 feefeffe 3ffefba0 40202636  
3fff0cc0:  feefeffe 3ffe84d8 3ffe84c4 4020279e  
3fff0cd0:  feefeffe feefeffe feefeffe feefeffe  
3fff0ce0:  feefeffe feefeffe feefeffe feefeffe  
3fff0cf0:  feefeffe feefeffe feefeffe feefeffe  
3fff0d00:  feefeffe feefeffe feefeffe feefeffe  
3fff0d10:  feefeffe feefeffe feefeffe feefeffe  
3fff0d20:  feefeffe feefeffe feefeffe feefeffe  
3fff0d30:  feefeffe feefeffe feefeffe feefeffe  
3fff0d40:  401052ce 00000000 3ffe8404 40205ab8  
3fff0d50:  00000020 40105427 3ffefdf0 3ffefc54  
3fff0d60:  40205a34 0001c200 4020674c 3ffe8477  
3fff0d70:  3ffefd1c 0000001a 3ffe8476 40206c90  
3fff0d80:  0000001c 3ffefd1c 3fff1118 3ffefddc  
3fff0d90:  3fffdad0 00000000 3ffefd1c 402060bd  
3fff0da0:  3ffefba0 00000000 00000001 feefeffe  
3fff0db0:  feefeffe feefeffe 3ffefd1c 3ffefddc  
3fff0dc0:  3fffdad0 3ffefba0 3ffefd1c 4020238c  
3fff0dd0:  feefeffe feefeffe feefeffe feefeffe  
3fff0de0:  feefeffe feefeffe feefeffe 3ffefddc  
3fff0df0:  3fffdad0 00000000 3ffefdd5 40206790  
3fff0e00:  feefeffe feefeffe 3ffefdf0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
system param error
0f 31 c8 ff c0 20 00 29 03 2d 0c 01 cb ff c0 00 00 08 31 c8 21 12 c1 10  €

```

Source code 

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>

const char* ssid = "xxxxxx";
//at least 8 characters long
const char* password = "xxxxxxxxx";

ESP8266WebServer server(80);
ESP8266HTTPUpdateServer httpUpdater;

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp12!");
  digitalWrite(led, 0);
}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  Serial.println();
  Serial.print("Configuring access point...");
  WiFi.begin(ssid, password);
    
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.on("/reset", [](){
    server.send(200, "test/plain", "ESP reset");
    delay(500);
    ESP.restart();
  });

  server.onNotFound(handleNotFound);

  httpUpdater.setup(&server);
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
}
```

**I use latest staging (2.1.0-rc2)**

Pins:
VCC, EN - **3.3V**
GPIO15,GND - **GND**

bblanchon, 2016-02-18:
Hi.
How is that related to ArduinoJson?

alinleonard, 2016-02-18:
wrong tab, sorry long hours awake

closed 2016-02-18 by bblanchon