bblanchon/ArduinoJson #228: Time_t compilation error

ianerito, 2016-02-19:
Dear bblanchon,
I am using your library of Json, and I am building a Json with one parameter of the type ot Time_t , one week ago I did this prefectly, but now, I am testing the same code with other board and all the lines work well except when I try to assign the value of T in createdAt, do you now how can I solve it?
Thanks for your time

`time_t T = now();
const int BUFFER_SIZE2 = JSON_OBJECT_SIZE(3);
StaticJsonBuffer<BUFFER_SIZE2> Buffer_json;
JsonObject& jsonValor = Buffer_json.createObject();
   jsonValor["senderUid"] = "84b";
   jsonValor["createdAt"] = T;
    jsonValor["value"] = "34";`

bblanchon, 2016-02-19:
Hi @ianerito,

I tried on my computer with the Time library which is in the Library Manager and it works fine.

If you have a compilation error, please copy/paste the message.
As `time_t` is not defined by Arduino itself, it comes from a library, do you know which one?

ianerito, 2016-02-19:
I think the problem is that I am working with the Arduino 101 with the Curie core, which must use the library CurieTime.h.

When I worked with the arduino micro with Time.h it works correctly

bblanchon, 2016-02-19:
CurieTime includes `time.h` from the standard library.
https://github.com/01org/corelibs-arduino101/blob/0c1ba8baef33e74383c0be797955baf386d5c4aa/libraries/CurieTime/src/CurieTime.cpp#L23

But the standard says that the content of `time_t` is implementation dependant:
http://en.cppreference.com/w/c/chrono/time_t

Once again, it would really help if you could just copy/paste the compiler error...

ianerito, 2016-02-19:
Here do you have all the response of the compiler, I did not send you this first because it is very long.
If I convert the variable T in a long, the compiler allows me to upload the sketch.
I have build the Json like I told you in the first message.
Thanks for your time.

```
Arduino:1.6.7 (Windows 10), Placa:"Arduino 101"

In file included from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObject.ipp:10:0,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObject.hpp:151,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson.h:9,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:13,

                 from C:\Users\PORTATILIMG\Documents\Arduino\COMPLETO\COMPLETO.ino:10:

C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp: In instantiation of 'ArduinoJson::JsonObjectSubscript<TKey>& ArduinoJson::JsonObjectSubscript<TKey>::operator=(const T&) [with T = long long int; TKey = const char*]':

C:\Users\PORTATILIMG\Documents\Arduino\COMPLETO\COMPLETO.ino:207:29:   required from here

C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp:35:25: error: conversion from 'const long long int' to 'ArduinoJson::JsonVariant' is ambiguous

         JsonVariant>(src);

                         ^

C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp:35:25: note: candidates are:

In file included from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonVariant.hpp:106:0,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:14,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:9,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/DynamicJsonBuffer.hpp:9,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson.h:7,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:13,

                 from C:\Users\PORTATILIMG\Documents\Arduino\COMPLETO\COMPLETO.ino:10:

C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonVariant.ipp:19:8: note: ArduinoJson::JsonVariant::JsonVariant(const char*) <near match>

 inline JsonVariant::JsonVariant(const char *value) {

        ^

C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonVariant.ipp:19:8: note:   no known conversion for argument 1 from 'const long long int' to 'const char*'

C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonVariant.ipp:93:8: note: ArduinoJson::JsonVariant::JsonVariant(short unsigned int)

 inline JsonVariant::JsonVariant(unsigned short value) {

        ^

C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonVariant.ipp:81:8: note: ArduinoJson::JsonVariant::JsonVariant(unsigned int)

 inline JsonVariant::JsonVariant(unsigned int value) {

        ^

C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonVariant.ipp:87:8: note: ArduinoJson::JsonVariant::JsonVariant(long unsigned int)

 inline JsonVariant::JsonVariant(unsigned long value) {

        ^

[... 65 lines cut ...]
   FORCE_INLINE TImpl& assign(TValue value) {

                       ^

In file included from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObject.ipp:10:0,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObject.hpp:151,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson.h:9,

                 from C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:13,

                 from C:\Users\PORTATILIMG\Documents\Arduino\COMPLETO\COMPLETO.ino:10:

C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp: In member function 'ArduinoJson::JsonObjectSubscript<TKey>& ArduinoJson::JsonObjectSubscript<TKey>::operator=(const T&) [with T = long long int; TKey = const char*]':

C:\Users\PORTATILIMG\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp:36:3: warning: control reaches end of non-void function [-Wreturn-type]

   }

   ^

exit status 1
Error de compilación
```

bblanchon, 2016-02-19:
Well, it's crystal clear:

```
error: conversion from 'const long long int' to 'ArduinoJson::JsonVariant' is ambiguous
```

On this platform `time_t` it's a `long long`.

So you have two options:

1. cast to a `long` as you did
2. enable support for `long long` in ArduinoJson (with flag `ARDUINOJSON_USE_LONG_LONG`)

Option 1 seems to be the way to go, because option 2 would increase the memory usage with no benefit.

ianerito, 2016-02-19:
Thank you for your answers!

bblanchon, 2016-02-19:
You're welcome, thanks for using ArduinoJson.

Don't hesitated to edit the wiki if you think you can improve it.
In particular, feel free to add you projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).
Also, a star on the project is always appreciated :wink:

closed 2016-02-19 by bblanchon