bblanchon/ArduinoJson #229: Write to JSON array

MCUdude, 2016-02-19:
Hi!
I'm trying to write to a JSON array I want to store a value to the first line of the array. My code looks like this

//Global`
StaticJsonBuffer<200> jsonDataBuffer; // 200 bytes JSON buffer for UART communication
JsonObject& jsonSerial = jsonDataBuffer.createObject(); // A reference to the JsonObject, the actual bytes are inside the JsonDataBuffer
JsonArray& Accelerometer1 = jsonSerial.createNestedArray("Accelerometer1"); // Object to create an array for Accelerometer 1

// In the setup
Accelerometer1.add(1); // Accelerometer 1; X-axis
Accelerometer1.add(2); // Accelerometer 1; Y-axis
Accelerometer1.add(3); // Accelerometer 1; Z-axis

// In the loop
jsonSerial["Accelerometer1"][0] = 13; //some numeric value
`

This gives me a really cryptic error message
`
JSON_test.ino: In function 'void loop()':
JSON_test:111: error: passing 'const ArduinoJson::JsonArraySubscript' as 'this' argument of 'TImpl& ArduinoJson::JsonSubscriptBase<TImpl>::operator=(int) [with TImpl = ArduinoJson::JsonArraySubscript]' discards qualifiers [-fpermissive]
passing 'const ArduinoJson::JsonArraySubscript' as 'this' argument of 'TImpl& ArduinoJson::JsonSubscriptBase<TImpl>::operator=(int) [with TImpl = ArduinoJson::JsonArraySubscript]' discards qualifiers [-fpermissive]`

How can I write to an array?

bblanchon, 2016-02-19:
```c++
Accelerometer1[0] = 13;
```

MCUdude, 2016-02-19:
Great! That did the trick. However, I couldn't find information on how to write to a place in the array anywhere in the examples, only read and create

bblanchon, 2016-02-19:
This is because, at compile time, we don't know that `jsonSerial["Accelerometer1"]` is an array.
But you could help the compiler by adding a cast:

```c++
jsonSerial["Accelerometer1"].as<JsonArray>()[0] = 13;
```

But it's awful, an performance is bad.

bblanchon, 2016-02-19:
Also, be careful when reusing the same `JsonBuffer`.
In this small snippet it's OK, but as soon as you do parsing you should not reuse the same `JsonBuffer`.

See [Avoiding pitfalls: Don't reuse the same JsonBuffer](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#4-dont-reuse-the-same-jsonbuffer) .

MCUdude, 2016-02-19:
Wouldn't this means that I'll loose the content of the Json file if I get rid of the buffer? My thought was that I could have a global json file, and be able to read and write to it when I wanted to.

bblanchon, 2016-02-19:
In a sense, yes.
As you can see you only keep references (ie pointers) to `JsonArray` and `JsonObject`, the actual bytes are located in the `JsonBuffer`.
When the `JsonBuffer` is discarded, the references become invalid.

You can read and write to the same `JsonBuffer` endlessly, as long as you reuse the same bytes, which is exactly what you do in the snippet above.
For example, if instead you were calling `JsonArray::add()`, it wouldn't work because the `JsonBuffer` would keep growing and growing.

MCUdude, 2016-02-23:
Great! Got it all working perfectly now :D

closed 2016-02-23 by MCUdude