bblanchon/ArduinoJson #230: Detect if object exist?

koffienl, 2016-02-20:
If read the wiki, but can't find a anwer. I'm feeling I'm overlooking something but can
t find out the correct way.
Let's say I have this simple json:

```
{
	"object1":
	{
		"key": "val"
	},
	"object2":
	{
		"key": "val"
	}
}
```

Now, I want to detect if `object2` exists at all, before writing to it.
When trying to find out it a object is empty or not:
```
JsonObject& obj = json[char_object];
if (obj.containsKey(char_item))
```

This would be false when the key doesn't exist *or* the entire object doesn't exist. How can I find out if the object even exist before doing a containsKey ?

bblanchon, 2016-02-20:
```c++
char json[] = "{\"object1\":{},\"object2\":{}}";
JsonObject& root = jsonBuffer.parseObject(json);
if (root.contains("object2")) {
   Serial.println("Found object2 !");
}
```

bblanchon, 2016-02-20:
You're welcome, thanks for using ArduinoJson.

Don't hesitated to edit the wiki if you think you can improve it.
In particular, feel free to add you projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).
Also, a GitHub star is always appreciated :wink:

closed 2016-02-20 by bblanchon