bblanchon/ArduinoJson #231: Returning JsonObject from a class function

hreintke, 2016-02-20:
LS,

I have a problem using the ArduinoJson within the Sming Framework
Probably a c++ issue, and if I missed some documentation please point me to that and I will work from there.

In the framework I have a class Command, which holds as a String a command which has entered the system
The Json part of it is : 
```
class Command
{
public:
	Command();
	Command(String reqString);
	Command(String reqName, String reqString);
	virtual ~Command();

	JsonObject& getRoot();

private :
	DynamicJsonBuffer *cmdJsonBuffer;
	JsonObject* cmdRoot;

	String cmdName = "";
	String cmdString = "";
};
```
I made cmdJsonBuffer and cmdRoot a pointer as the cmdString might not be complete at instantiation.
Also that will save memory when the application knows it is not a json string and will not try to access it that way.

But for the ones which want to,  I want to make an interface to access the command as jsonObject.

```
JsonObject& Command::getRoot()
{
	if (!cmdJsonBuffer)
	{
		cmdJsonBuffer = new DynamicJsonBuffer;
		cmdRoot =  &cmdJsonBuffer->parseObject(cmdString);
	}
	return *cmdRoot;
}
```
I cannot do the parseObject at instantiation as the contents os cmdString is not known/valid at that time.

The application then is : 
```
void processTestCommands(Command reqCommand, CommandOutput* commandOutput)
{
	JsonObject& root = reqCommand.getRoot();
	if (!root.success())
	{
		commandOutput->printf("Not a valid json command\r\n");
	}
	else
	{
		// process the json command
	}
}
```

The issue is at the root.success(). I always get an exception.
This also happens when I test within the command class
```
	if (cmdRoot->success()) {debugf("getRoot success");}
	return *cmdRoot;
```

Can you point me to the error I am making or an example which is showing the functionality I am implementing ?

bblanchon, 2016-02-21:
Hi,

I read you question carefully several times, but unfortunately I don't understand what you want to do, what you're actually doing and what is the problem that you're having.

All I can say is: if you have an exception when calling `JsonObject::success()` then you probably have a NULL pointer or a dangling pointer.
It's not obvious from the code you show why it's like this.

If you need more help, please rephrase the question and try to focus your question on the essential part of the issue.

Regards,
Benoit

PS: what means LS?

hreintke, 2016-02-22:
Ok, trying to me mmore clear about my question.

First on the LS, it's a (little old-fashioned I know) way of saying "Hello to the reader". It an abbreviation from the latin "Lectori Salutem" and originates from the Romans.

Then the json question.

What I like to achieve is that the class can be used in such a way that

f.e. in the application :
```
Command myCommand;
...
JsonObject& myRoot = myCommand.getRoot();
String value = myRoot["test"];
...
```
That hides the use of Static/Dynamic jsonbuffer and createObject(...) within the class.

The issue I have is that within the class I can have :
```
DynamicJsonBuffer jsonBuffer;
JsonObject &classJson; 
```
-> Requires an object at instantiation, which I can do with classObject = jsobBuffer.createObject(); (The String with the input is not yet available)
-> But then I cannot add objects/elemements from a string (like I would do with parseObject(String))
-> And I cannot do another classObject = parseObject(String) -> error: use of deleted function

So I decided to go for pointers in the class so I could instantiate the jsonBuffer & jsonObject when needed
At that point in time, also the String containing input is available.
```
DynamicJsonBuffer* jsonBuffer;
JsonObject* classObject ;
```
And when needed I do : 
```
jsonBuffer = new DynamicJsonBuffer;
classJson =  &jsonBuffer->parseObject(cmdString);
```
I expect(ed) that classJson would hold the pointer to the newly instantiated jsonObject.
Returning the object itself as function result : return *classJson;

I checked the (void *) value of classJson and that is not NULL.
But when I use the `classJson->success()` after the `classJson =  &jsonBuffer->parseObject(cmdString);`for checking the result I get exception.

Hope is is some more clear now.
As mentioned in my first post, I don't know whether it is a c++ or ArduinoJson issue but probably I am "solution blind" and would like someone with knowledge from both, not to solve but just a hint in the right direction.

bblanchon, 2016-02-22:
I think I now understand what you want to do.

I can confirm that `classJson =  &jsonBuffer->parseObject(cmdString);` does exactly what you think it does.

You have an exception because `classJson` is dangling, most likely because the `DynamicJsonBuffer` has been destroyed.

One explanation could be that the instance of `Command` is copied.
Indeed the default copy constructor will blindly copy the pointers, so when the original object is deleted, the pointers starts to dangle.

To verify this hypothesis, just delete the copy constructor:

```
class Command
{
public:
    Command(const Command&) = delete;

    //...
};
```

I you try to make a copy of Command, then a compiler error will occur.

hreintke, 2016-02-22:
Hi,
Sorry for spending your time for a starters error I made. 
When checking whether a pointer is already filled `if (!cmdJsonBuffer)`
You better make sure that it is initialized to NULL.. 

I found the mistake when making the example even smaller to 
```
	if (!cmdJsonBuffer)
	{
		debugf("Filling cmdRoot");
		cmdJsonBuffer = new DynamicJsonBuffer;
		cmdRoot =  &cmdJsonBuffer->parseObject(cmdString);
		if (cmdRoot->success()) { .....} else {...}
	}
```

Thanks a lot for helping out, .

PS : I took the `Command(const Command&) = delete;` advise from you to prevent accidental copy of the objects.

bblanchon, 2016-02-22:
You're welcome, thanks for using ArduinoJson.

Don't hesitated to edit the wiki if you think you can improve it.
In particular, feel free to add you projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).
Also, a GitHub star is always appreciated :wink:

closed 2016-02-22 by bblanchon

hreintke, 2016-02-22:
Hi,
Quick followup question.
Is the jsonObject destructed when jsonBuffer is destructed ? 
When having :
`JsonObject* cmdRoot `
And later : `delete cmdRoot`
I get : 
```
sming/commandprocessing/Command.cpp:39:9: error: no suitable 'operator delete' for 'ArduinoJson::JsonObject'
  delete cmdRoot;
```