bblanchon/ArduinoJson #234: Issue

aslamy, 2016-02-26:
Board crashes with this settings.

const unsigned long BUFFER_SIZE = JSON_OBJECT_SIZE(2) + JSON_ARRAY_SIZE(1000);
StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;

bblanchon, 2016-02-26:
This is totally normal, it's a stack overflow.
You're just allocating more memory than what's available.
For example, on the ESP8266, I think that the stack size is limited to 4KB.

To solve this, either:

1. reduce the size of the `StaticJsonBuffer`
2. use a `DynamicJsonBuffer`, because the memory is allocated on the heap
3. buy a bigger MCU with more RAM

aslamy, 2016-02-26:
it menas:
 BUFFER_SIZE = 12048;
StaticJsonBuffer<12048> jsonBuffer

and board crashing but when i have for example:
Static Json Buffer<20000> jsonBuffer;
which is much more the board not crashing.

bblanchon, 2016-02-26:
To be more exact, allocating memory doesn't make the device crash.
Usualy, what makes the device crash is when the memory is written, because it will override other variables.
So depending on the positions of variables in memory it may work, but it's a bomb waiting to explode. :bomb:

aslamy, 2016-02-26:
I have tried: 
```
StaticJsonBuffer<20000> jsonBuffer;
StaticJsonBuffer<40000> jsonBuffer;
StaticJsonBuffer<60000> jsonBuffer;
StaticJsonBuffer<80000> jsonBuffer;
```
and everything is ok.  Its just some size like 12048 or 12068 which causing crash.

bblanchon, 2016-02-26:
As explained in my previous message: the behavior is random, as it depends on the positions of variables in memory.
It's always like that with stack overflows.

Did you try the suggested workaround?

aslamy, 2016-02-26:
I use a board called linkit one and i have not that limitation of ram. It has 4MB of ram.

bblanchon, 2016-02-26:
4MB of RAM doesn't mean 4MB of stack.

Did you try the suggested workaround:question:

aslamy, 2016-02-26:
DynamicJsonBuffer Working create. How much slower is it?

bblanchon, 2016-02-26:
Compared to a device that crashes, I would say that's the speed boost is +infinity :-)

Seriously, if you specify the expected size in the constructor, instead of letting the buffer grow by itself, the speed difference is insignificant.

Also, please keep in mind that ArduinoJson is not the fastest JSON library.
The goal of ArduinoJson is to be able to run on microcontrollers with very limited resources.
It performs well on bigger platforms, but other library are suitable too.

aslamy, 2016-02-26:
hahaaha ok thank you !!

closed 2016-02-26 by aslamy

bblanchon, 2016-02-26:
You're welcome, thanks for using ArduinoJson.

Don't hesitated to edit the wiki if you think you can improve it.
In particular, feel free to add you projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).
Also, a GitHub star is always appreciated :wink: