bblanchon/ArduinoJson #235: What is performance between DynamicJsonBuffer and  StaticJsonBuffer object ?

aslamy, 2016-02-26:
What is the performance difference between:

```
DynamicJsonBuffer jsonBuffer ;
Some code...
```

and 

```
StaticJsonBuffer<2000> *jsonBuffer = new StaticJsonBuffer<2000>();
Some code...
delete(jsonBuffer);
```

bblanchon, 2016-02-27:
The default initial size for `DynamicJsonBuffer` is 256.
Each time it needs to grow, it allocates a new block twice bigger than the previous one.
To to reach 2000 bytes:

1. First block: 256
2. Second block: 512 (total 768)
3. Third block: 1024 (total 1792)
4. Fourth block: 2048

So `DynamicJsonBuffer` will have to make 4 allocations instead of only 1 for `StaticJsonBuffer`.

However, if you specify the size of the `DynamicJsonBuffer` to its constructor, you can have the exact same performance:

```c++
DynamicJsonBuffer jsonBuffer(2000);
```

Also, once again, the focus of ArduinoJson is not on speed!
The goal of ArduinoJson is to provide a JSON library that can work reliably on devices with low resources.

aslamy, 2016-02-27:
OK thanks . But I think ArduinoJson library works perfectly in terms of memory consumption.I think it is time to find solutions to increase performance. I have tested aJson. It has better performance but are very bad in terms of memory consumption.

bblanchon, 2016-02-27:
Thank you for this advice, but compromises will always be made in favor of code size, then memory consumption and then speed.

Also, before claiming that ArduinoJson has poor performance, please provide solid facts.
In particular, please prove that aJson is significantly faster than ArduinoJson on an embedded platform.

closed 2016-03-04 by bblanchon