bblanchon/ArduinoJson #236: Parsing array with keys values

androme13, 2016-02-27:
Hi, i try to parse something like that :
{"action":"CBL","parameters":"[{\"Push_Type\":\"0\",\"Role\":\"1\",\"Depend\":\"0\",\"Timer_On\":\"0\",\"Action_Key\":\"\",\"Repeat_Delay\":\"100\",\"Led_Type\":\"0\",\"Led_Blink_Delay\":\"100\",\"Relay\":\"0\"},{\"Push_Type\":\"0\",\"Role\":\"0\",\"Depend\":\"0\",\"Timer_On\":\"0\",\"Action_Key\":\"\",\"Repeat_Delay\":\"100\",\"Led_Type\":\"0\",\"Led_Blink_Delay\":\"100\",\"Relay\":\"0\"},{\"Push_Type\":\"0\",\"Role\":\"0\",\"Depend\":\"0\",\"Timer_On\":\"0\",\"Action_Key\":\"\",\"Repeat_Delay\":\"100\",\"Led_Type\":\"0\",\"Led_Blink_Delay\":\"100\",\"Relay\":\"0\"},{\"Push_Type\":\"0\",\"Role\":\"0\",\"Depend\":\"0\",\"Timer_On\":\"0\",\"Action_Key\":\"\",\"Repeat_Delay\":\"100\",\"Led_Type\":\"0\",\"Led_Blink_Delay\":\"100\",\"Relay\":\"0\"},{\"Push_Type\":\"0\",\"Role\":\"0\",\"Depend\":\"0\",\"Timer_On\":\"0\",\"Action_Key\":\"\",\"Repeat_Delay\":\"100\",\"Led_Type\":\"0\",\"Led_Blink_Delay\":\"100\",\"Relay\":\"0\"},{\"Push_Type\":\"0\",\"Role\":\"0\",\"Depend\":\"0\",\"Timer_On\":\"0\",\"Action_Key\":\"\",\"Repeat_Delay\":\"100\",\"Led_Type\":\"0\",\"Led_Blink_Delay\":\"100\",\"Relay\":\"0\"},{\"Push_Type\":\"0\",\"Role\":\"0\",\"Depend\":\"0\",\"Timer_On\":\"0\",\"Action_Key\":\"\",\"Repeat_Delay\":\"100\",\"Led_Type\":\"0\",\"Led_Blink_Delay\":\"100\",\"Relay\":\"0\"},{\"Push_Type\":\"0\",\"Role\":\"0\",\"Depend\":\"0\",\"Timer_On\":\"0\",\"Action_Key\":\"\",\"Repeat_Delay\":\"100\",\"Led_Type\":\"0\",\"Led_Blink_Delay\":\"100\",\"Relay\":\"0\"},{\"Push_Type\":\"0\",\"Role\":\"0\",\"Depend\":\"0\",\"Timer_On\":\"0\",\"Action_Key\":\"\",\"Repeat_Delay\":\"100\",\"Led_Type\":\"0\",\"Led_Blink_Delay\":\"100\",\"Relay\":\"0\"},{\"Push_Type\":\"0\",\"Role\":\"0\",\"Depend\":\"0\",\"Timer_On\":\"0\",\"Action_Key\":\"\",\"Repeat_Delay\":\"100\",\"Led_Type\":\"0\",\"Led_Blink_Delay\":\"100\",\"Relay\":\"0\"}]"}
Is there a way to decode the array by keys and values ?
I'm stuck.

bblanchon, 2016-02-27:
Hi,
What is the problem?
Have you read the [FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ)?

androme13, 2016-02-28:
i tried that :
```
StaticJsonBuffer<2000> jsonBufferIn;
JsonObject& rootIn = jsonBufferIn.parseObject(str);
String action = rootIn["action"];
 if (action == "CBL") {
        Serial.println("load profile");
        String parameters = rootIn["parameters"];
        Serial.println(parameters); // parameters string is ok and filled
        StaticJsonBuffer<2000> jsonBufferArray;
        JsonArray& array2 = jsonBufferArray.parseArray(parameters);
        Serial.println(array2.size()); // = 0 here
    }
```
I think i don't have the good way.

bblanchon, 2016-02-28:
The code seems correct, but it seems that you're looking trouble:

1. Why including a JSON string in a JSON object ([aka Xzibit style](http://cdn.meme.am/instances/500x/59375925.jpg))?
2. Why using `String` everywhere?

Here is, what I would do if I were you:

1 -> Change the JSON to:

```json
{
  "action": "CBL",
  "parameters": [
    {
      "Push_Type": 0,
      "Role": "1",
      "Depend": 0,
      "Timer_On": 0,
      "Action_Key": "",
      "Repeat_Delay": 100,
      "Led_Type": 0,
      "Led_Blink_Delay": 100,
      "Relay": 0
    },
    {
      "Push_Type": 0,
      "Role": 0,
      "Depend": 0,
      "Timer_On": 0,
      "Action_Key": "",
      "Repeat_Delay": 100,
      "Led_Type": 0,
      "Led_Blink_Delay": 100,
      "Relay": 0
    }
  ]
}
```

2 -> Change the program to:

```c++
StaticJsonBuffer<2000> jsonBuffer;
JsonObject& root = jsonBuffer.parseObject(str);

const char* action = root["action"];
if (!strcmp(action, "CBL")) {
    JsonArray& parameters = root["parameters"];
    Serial.println(parameters.size()); 
    // ...
}
```

Voila!

androme13, 2016-02-28:
if my json is : 
`{"action":"CBL","parameters":[{"Push_Type":"0","Role":"1","Depend":"0","Timer_On":"0","Action_Key":"","Repeat_Delay":"100","Led_Type":"0","Led_Blink_Delay":"100","Relay":"0"},{"Push_Type":"0","Role":"0","Depend":"0","Timer_On":"0","Action_Key":"","Repeat_Delay":"100","Led_Type":"0","Led_Blink_Delay":"100","Relay":"0"}]}`

even the action cannot be decoded.

androme13, 2016-02-28:
I have somes news, my array contains ten "objects", if i use` StaticJsonBuffer<2000> jsonBuffer`, it fails, but il i use `DynamicJsonBuffer jsonBufferIn`, it's ok
the whole json string is 1481 chars so less than 2000 declared in StaticJsonBuffer;

bblanchon, 2016-02-28:
Usually the `JsonBuffer` don't have to make a copy of the JSON string.
It only does it if the input is const.
You should use a `char[]` or cast the input.

If the input is 1481, it leaves only 518 for the JSON object tree.
This seems OK, but maybe that's not enough.

Also, maybe there is a stackoverflow.
I don't know if the stack size is limited on Arduino DUE, but that would be a good explanation.
Anyway, since it has 96KB of RAM, it's perfectly acceptable to use `DynamicJsonBuffer`.

You may want to read [Wiki: Memory model](https://github.com/bblanchon/ArduinoJson/wiki/Memory-model).

androme13, 2016-02-28:
I'll will do as you advise, anyway thank you.

closed 2016-02-28 by androme13

bblanchon, 2016-02-28:
De rien, merci d'utiliser ArduinoJson.

N'hésite pas à éditer le wiki si tu penses qu'il y a des choses à améliorer.
En particulier, tu peux ajouter ton projet dans [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).
Et puis, une étoile GitHub star fera toujours plaisir :wink: