bblanchon/ArduinoJson #237: Problem to Parse

gustavogini, 2016-02-27:
Hello bblanchon!

I`m assemblyng a tilt and pan controller. I already encoding the information with json but I`m facing problems to receive the information and decoding it. 

I`m trying to communicate two arduino with ethernet shields in a local network.

**Server code:**
`void loop() {
    // listen for incoming clients
  EthernetClient client = server.available();
  if (client) {
    Serial.println("new client");
    // an http request ends with a blank line
    boolean currentLineIsBlank = true;
    while (client.connected()) {
      if (client.available()) {
        char c = client.read();
        Serial.write(c);
        // if you've gotten to the end of the line (received a newline
        // character) and the line is blank, the http request has ended,
        // so you can send a reply
        if (c == '\n' && currentLineIsBlank) {
          // send a standard http response header
          client.println("HTTP/1.1 200 OK");
          client.println("Content-Type: text/html");
          client.println("Connection: close");  // the connection will be closed after completion of the response
          client.println("Refresh: 5");  // refresh the page automatically every 5 sec
          client.println();
  
  jtkPan = analogRead (joystickPanPin);
  jtkTilt = analogRead (joystickTiltPin);
  jtkZoom = analogRead (joystickZoomPin);
  
  potPan = analogRead (PotencPan);
  potTilt = analogRead (PotencTilt);
  potZoom = analogRead (PotencZoom);
  
  jtkPanPos = funcPan1 (jtkPan);
  jtkTiltPos = funcTilt1 (jtkTilt);
  jtkZoomPos = funcZoom1 (jtkZoom);
  
  //Serial.print(jtkPanPos);
  //Serial.print("  ");
  //Serial.print(jtkTiltPos);
  //Serial.print("  ");
  //Serial.print(jtkZoomPos);
  //Serial.println ("  ");
  
  
  VelocHorPan = funcAumentarVelocSentidoHorarioPan (potPan);
  VelocAntiHorPan = funcAumentarVelocSentidoAntiHorarioPan (potPan);
  
  VelocHorTilt = funcAumentarVelocSentidoHorarioTilt (potTilt);
  VelocAntiHorTilt = funcAumentarVelocSentidoAntiHorarioTilt (potTilt);
  
  VelocHorZoom = funcAumentarVelocSentidoHorarioZoom (potZoom);
  VelocAntiHorZoom = funcAumentarVelocSentidoAntiHorarioZoom (potZoom);
  
  
  char jspan[10];
  char jstilt[10];
  char jszoom[10];
  
   
  ultoa(funcPan2(jtkPanPos, VelocHorPan, VelocAntiHorPan),jspan,10);
  ultoa(funcTilt2(jtkTiltPos, VelocHorTilt, VelocAntiHorTilt),jstilt,10);
  ultoa(funcZoom2(jtkZoomPos, VelocHorZoom, VelocAntiHorZoom),jszoom,10);
  
  //a volta de string para long int eh atol
  
  //char jspan = (char) funcPan2(jtkPanPos, VelocHorPan, VelocAntiHorPan);
  //char jstilt = (char) funcTilt2(jtkTiltPos, VelocHorTilt, VelocAntiHorTilt);
 // char jszoom = (char) funcZoom2(jtkZoomPos, VelocHorZoom, VelocAntiHorZoom);
  
  
   StaticJsonBuffer<200> jsonBuffer;
JsonObject& root = jsonBuffer.createObject();

root["Pan"] = jspan;
Serial.println(jspan);
root["Tilt"] = jstilt;
Serial.println(jstilt);
root["Zoom"] = jszoom;
Serial.println(jszoom);

root.printTo(client);
          
          
          break;
        }
        if (c == '\n') {
          // you're starting a new line
          currentLineIsBlank = true;
        }
        else if (c != '\r') {
          // you've gotten a character on the current line
          currentLineIsBlank = false;
        }
      }
    }
    // give the web browser time to receive the data
    delay(1);
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }

delay(1000);
}`

**Web Client Code**: `

#include <SPI.h>
#include <Ethernet.h>
#include <ArduinoJson.h>
#include <stdlib.h>

// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xEB };
// if you don't want to use DNS (and reduce your sketch size)
// use the numeric IP instead of the name for the server:
IPAddress server(192,168,2,10);  // IP servidor

// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 2, 11);

// Initialize the Ethernet client library
// with the IP address and port of the server
// that you want to connect to (port 80 is default for HTTP):
EthernetClient client;

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  // start the Ethernet connection:
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    // no point in carrying on, so do nothing forevermore:
    // try to congifure using IP address instead of DHCP:
    Ethernet.begin(mac, ip);
  }
  // give the Ethernet shield a second to initialize:
  delay(1000);
  Serial.println("connecting...");

  // if you get a connection, report back via serial:
  if (client.connect(server,80)) {
    Serial.println("connected");
    // Make a HTTP request:
    client.println("GET /search?q=arduino HTTP/1.1");
    client.println("Connection: close");
    client.println();
  }
  else {
    // kf you didn't get a connection to the server:
    Serial.println("connection failed");
  }
}

void loop()
{
  // if there are incoming bytes available
  // from the server, read them and print them:
  if (client.available()) {
    char jsonBuf[2000];
    char json;
   
    json = client.read();
     //utoa(json,jsonBuf,10);
    Serial.print(jsonBuf);
    
   StaticJsonBuffer<2000> jsonBuffer;
   JsonObject& root = jsonBuffer.parseObject(jsonBuf);
	
	if (!root.success()){
    Serial.println("parseObject() failed");
    return;
   }
   
    int panjson = root["pan"];
    int tiltjson = root["tilt"];
    int zoomjson = root["zoom"];
	
	Serial.println(panjson);
	Serial.println(tiltjson);
	Serial.println(zoomjson);
	
  }

  // if the server's disconnected, stop the client:
  if (!client.connected()) {
    Serial.println();
    Serial.println("disconnecting.");
    client.stop();

    // do nothing forevermore:
    while (true);
  }
}`

Could you please help me? I can`t identify the code problems.

Thanks in advance.

bblanchon, 2016-02-28:
Hello Gustavo,

That's a lot of code :astonished: 
Can you please extract the relevant part?

With a quick glance, I saw something disturbing...
You say that you work on an Arduino with Ethernet shield, right?
You don't mention the model of the board, but I think it's fair to assume that it's a UNO.
The Arduino UNO only has 2KB of RAM.
So `char jsonBuf[2000];` and `StaticJsonBuffer<2000> jsonBuffer;` will never fit.
At best they could be 512 bytes each, leaving 1024 for the rest of the program, but that may still be too much because there is the Ethernet library that is using a lot of memory too.

Regards,
Benoit

gustavogini, 2016-02-28:
Hello Benoit,

Sorry for the amount of code.
I am using two arduino uno with W5100 ethernet shields.
I changed to 512bytes but I`m facing other problems to compile the decoding code.

void loop()
{
  if (client.available()) {
    char jsonBuf[512];
    char json;
   
    jsonBuf = client.read();
    Serial.print(jsonBuf);
    
   StaticJsonBuffer<512> jsonBuffer;
   JsonObject& root = jsonBuffer.parseObject(jsonBuf);
	
	if (!root.success()){
    Serial.println("parseObject() failed");
    return;
   }
    int panjson = root["pan"];
    int tiltjson = root["tilt"];
    int zoomjson = root["zoom"];
	
	Serial.println(panjson);
	Serial.println(tiltjson);
	Serial.println(zoomjson);	
  }

After compile this code with arduino software, it shows the following error "webClientJson:62: error: incompatible types in assignment of 'int' to 'char [512]'

Do you suggest any correction in the code? 

Thanks in advance

Best Regards
Gustavo

bblanchon, 2016-02-28:
That's because you need to call `client.readBytes()` or `client.readBytesUntil()`
https://www.arduino.cc/en/Reference/StreamReadBytes
https://www.arduino.cc/en/Reference/StreamReadBytesUntil

gustavogini, 2016-02-28:
Hello Benoit!

I changed to client.readBytes() and it works very well! Now I can receive the information and print in serial monitor to check.

After I receive the right information ({"Pan":"1600","Tilt":"1500","Zoom":"1500"}), the message comes with a lot of trash and after that the parse failed.

My code request to take each information and print in the serial monitor but It doesn`t happened because the parse failed.

Serial Monitor information:

connecting...
connected
HTTP/1.1 200 OK
Content-Type: text/html
Connection: close
Refresh: 5

{"Pan":"1600","Tilt":"1500","Zoom":"1500"}ûyýÞíç÷¾?\˜ó¯/­zÿ}¾½þçÜìÿö»},ÿË9xÇ~÷âÓX_úu?­û»Ó]|ÿ—eëýï“ëÑÌ±6¯ÿÑŒ¿ú…Rsg~t½—ôùæyýyØÝ¯õ£ïßAØ¿6©‹»Û1·vC¯^õ~óìjûqû_÷Â&þýcò;�¼Öû2âÿ2Æ¿¹ÿ=�wã®öÛýïÿÇ“¬Øçí¸ÕžÚÛyÓ«ý=+¦ëÿú@ÿûÛ<1/žM¿së«=ãü˜–­ÃÑ$‰6oÆË¨ÏÎ¸ªsuÃÞÈa\ù4]parseObject() failed

disconnecting.

gustavogini, 2016-02-28:
Hello Benoit,

I changed the 512bytes to 256bytes and the trash mentioned before dispeared but still apearing the parseObject() failed.

connecting...
connected
HTTP/1.1 200 OK
Content-Type: text/html
Connection: close
Refresh: 5

{"Pan":"1600","Tilt":"1500","Zoom":"1500"}PparseObject() failed

disconnecting.

What do you think about this problem?

That`s the full arduino client code.

Thanks in advance!

```
#include <SPI.h>
#include <Ethernet.h>
#include <ArduinoJson.h>
#include <stdlib.h>

// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xEB };
// if you don't want to use DNS (and reduce your sketch size)
// use the numeric IP instead of the name for the server:
IPAddress server(192,168,2,10);  // IP servidor

// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 2, 11);

// Initialize the Ethernet client library
// with the IP address and port of the server
// that you want to connect to (port 80 is default for HTTP):
EthernetClient client;

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  // start the Ethernet connection:
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    // no point in carrying on, so do nothing forevermore:
    // try to congifure using IP address instead of DHCP:
    Ethernet.begin(mac, ip);
  }
  // give the Ethernet shield a second to initialize:
  delay(1000);
  Serial.println("connecting...");

  // if you get a connection, report back via serial:
  if (client.connect(server,80)) {
    Serial.println("connected");
    // Make a HTTP request:
    client.println("GET /search?q=arduino HTTP/1.1");
    client.println("Connection: close");
    client.println();
  }
  else {
    // kf you didn't get a connection to the server:
    Serial.println("connection failed");
  }
}

void loop()
{
  // if there are incoming bytes available
  // from the server, read them and print them:
  if (client.available()) {
    char json[256];
    //char json;
   
    client.readBytes(json,256);
     //utoa(json,jsonBuf,10);
    Serial.print(json);
    
   StaticJsonBuffer<256> jsonBuffer;
   JsonObject& root = jsonBuffer.parseObject(json);
	
	if (!root.success()){
    Serial.println("parseObject() failed");
    return;
   }
   
    int panjson = root["Pan"];
    int tiltjson = root["Tilt"];
    int zoomjson = root["Zoom"];
	
	Serial.println(panjson);
	Serial.println(tiltjson);
	Serial.println(zoomjson);
  
	
  }

  // if the server's disconnected, stop the client:
  if (!client.connected()) {
    Serial.println();
    Serial.println("disconnecting.");
    client.stop();

    // do nothing forevermore:
    while (true);
  }
}
```

bblanchon, 2016-02-28:
### 1. About the garbage...

I guess that `readBytes()` doesn't add the zero-terminator.

Please try this:

```c++
size_t n = client.readBytes(json,256);
json[n] = 0;
```

This should remove the garbage at the end of the string.

But, you may want to use `readBytesUntil()` to read line by line... see below.

### 2. About the JSON payload...

ArduinoJson is only able to decode the JSON part.
So you need to skip the HTTP header.
It's quite easy, you just need to read lines one by one until you find an empty line.

See what I did in the RFID terminal:
https://github.com/drinks-wallet/DrinksRfidTerminal/blob/master/HttpClient.cpp#L129

```c++
// skip HTTP headers
while (readln(content, maxContentSize), NOT_EMPTY(content));

// read content
readln(content, maxContentSize);
```

BTW, thank you very much for the donation, it's well appreciated :beers:

gustavogini, 2016-02-29:
Hello Benoit!

As you mentioned in part 1, the garbage disapeard.

In the part 2 (Json payload) I removed the header of the http from my server and voilà, the parse works very well.

Thanks a lot for the Json and help me and the people!

Best Regards!

Gustavo Gini

closed 2016-02-29 by gustavogini