bblanchon/ArduinoJson #239: Number of decimal places for double

tombarks, 2016-02-28:
I'm trying to send ms accurate time from arduino to client. Using code:

root["Time"] = (double)millis() / 1000.0;

But this truncates to two decimal places and I need ms accurate time. There seems to be a way to set the number of decimals in an array but can't see how for a root object.

bblanchon, 2016-02-28:
Either:

```c++
double time = millis() / 1000.0;
uint8_t digits = 6;
root.set("Time", time, digits);
```

or

```c++
double time = millis() / 1000.0;
uint8_t digits = 6;
root["Time"] = double_with_n_digits(time, 6);
```

tombarks, 2016-02-28:
How about just:

root["Time"] = double_with_n_digits(((double)millis() / 1000.0), 3);

I only need 3 decimal places I think? Do I need to cast millis() as double here? Perhaps it will automagically do this for me but I am never quite sure of the behaviour.

bblanchon, 2016-02-28:
The result of `millis() / 1000.0` already has the type `double`, since `1000.0` is a double.

tombarks, 2016-02-28:
Thanks for your help and for making a great lib!

Tom

bblanchon, 2016-02-28:
You're welcome, thanks for using ArduinoJson.

Don't hesitated to edit the wiki if you think you can improve it.
In particular, feel free to add you projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).
Also, a GitHub star is always appreciated :wink:

closed 2016-02-28 by bblanchon