bblanchon/ArduinoJson #240: Create an array of Json objects

Vulcan300, 2016-03-03:
First of all, thank you for your work. And the second is to ask you about how to create an array of JSON objects. My program is to collect JSON objects which return a API. I can retrieve JSON objects and parse using you library but now I want to create an array with these objects.

Code:

      JsonArray& array = jsonBuffer.createArray();
      JsonObject& root = jsonBuffer.parseObject(bufferJson);

      if (!root.success()) 
      {
        Serial.println("parseObject() failed");
        return;
      }
      
      String AddDate = root["AddDate"];
      int MessageType = root["MessageType"];
      String ShortText = root["ShortText"];

      //This section prints the object fields perfectly
      Serial.print("AddDate: ");
      Serial.println(AddDate);
      Serial.print("MessageType: ");
      Serial.println(MessageType);
      Serial.print("ShortText: ");
      Serial.println(ShortText);
      
      array.add(root);

I run this code in a loop. After leaving the loop I print the array but only following it is shown:

AddDate: 2016-03-03T17:49:29.387
MessageType: 4
ShortText: TEMP. AMBIENTE
03/03/16 17:49
22.27 ÂºC

AddDate: 2016-03-03T17:53:42.19
MessageType: 4
ShortText: TEMP. AMBIENTE Habit
03/03/16 17:53
35.01 ÂºC

AddDate: 2016-03-03T17:57:01.07
MessageType: 4
ShortText: TEMP. AMBIENTE
03/03/16 17:57
21.78 ÂºC

Array:
[{"":"","":,"":"","":,"":},{"":"","":,"":"","":,"":},{"":"","":,"":"","":,"":}]

What am I doing wrong?

Thanks.

bblanchon, 2016-03-04:
Hi,

This part of the code looks good.
The problem probably comes from the rest of the program.

Most likely, it's because the input JSON string is not in memory anymore.
A quick fix would be to copy the input in the `JsonBuffer`, using `strdup()`:

```c++
DynamicJsonBuffer jsonBuffer; // <- caution will be big
const char* urls[] = {
   "http://myhome/sensors/1",
   "http://myhome/sensors/2",
   "http://myhome/sensors/3"
};

JsonArray& array = jsonBuffer.createArray();

for (int i=0; i<3; i++) {
    char json[256]; // <- local to the loop, will be discarded
    downloadJsonFrom(urls[i], json, sizeof(json));

    const char* jsonDuplicate = jsonBuffer.strdup(json); // <- make a copy in the JsonBuffer
    array.add(jsonBuffer.parseObject(jsonDuplicate));
}

array.printTo(Serial);
```

Caution: big JSON means big memory consumption, make sure you MCU can handle it.

Please have a look at:

- [ArduinoJson: Avoiding pitfalls](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls)
- [ArduinoJson: FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ)

Vulcan300, 2016-03-07:
Thanks, it works. :)

bblanchon, 2016-03-08:
You're welcome @Vulcan300, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add you projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-03-08 by bblanchon