bblanchon/ArduinoJson #242: Why is it unreasonable to have a resettable StaticJSONObject?

psigen, 2016-03-10:
I understand the justification that scoped construction of the `StaticJSONObject` avoids the unnecessary `clear()` semantics that you are very against here: https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-reuse-a-jsonbuffer 

But it seems like it would be very useful to be able to store the object in the [`.data`](https://en.wikipedia.org/wiki/Data_segment) segment and _not on the stack or heap at all_.  It makes it clear what the memory footprint is, avoids large swings in the stack size, and lets the JSON object be easily shared with interrupts and other out-of-order events.  Most of the other fixed-memory JSON parsers e.g. `jsmn` do this with their token arrays.

Is there some reason to disavow this use case?

bblanchon, 2016-03-10:
I designed this interface to be:

1. easy to get right
2. difficult to get wrong

I think the FAQ is quite clear on the reasons why it would be bad to have that feature.

If you think that global variable is the way to go, why don't you use the suggested workaround ?

psigen, 2016-03-10:
I am currently using the workaround, I was just wondering why it was necessary. After all, one can still always make memory errors like:

```c++
JsonObject& makeAnObject()
{
   StaticJSONBuffer<200> jsonBuffer;
   return jsonBuffer.createObject();
}
```

From my perspective, it is very semantically reasonable to want to do the following when parsing:

```c++
StaticJSONBuffer<200> buffer;
char line[256];

void main() {
    while(true) {
        line = readline();
        buffer.parseObject(line);  // <- buffer is internally cleared here.
        // Do some stuff
    }
}
```
This is the semantic used by many other parsing libraries, because the `parse()` function clears the buffer and stores only the resulting tokens.

But it is more complicated if you are allowing interleaved encoding and decoding operations to share the same buffer, as it may contain many top-level objects that should not be cleared.  So I can see why it would be simpler for you to simply avoid this use case.

bblanchon, 2016-03-10:
When you read this:

```c++
JsonObject& makeAnObject()
{
   StaticJsonBuffer<200> jsonBuffer;
   return jsonBuffer.createObject();
}
```

the mistake is quite visible.
Even if you don't know about reference, after a minute of googling "returning a C++ reference" you'll understand that returning a reference can be wrong.
Also it's a very common beginner mistake, that everyone make once, then understand.

On the other hand, when you read this:

```c++
StaticJsonBuffer<200> jsonBuffer;
JsonObject& object = jsonBuffer.createObject();
jsonBuffer.clear();
```

It's absolutely not obvious that `object` is a dangling reference.
In this case, it's not a classic beginner mistake, it won't be easy to find the answer.

Putting a `clear()` method in `JsonBuffer` would be an invitation  to make such mistake.
Remember that Arduino is a platform for beginner, so ArduinoJson has to respect that.

Finally, about your last remark: yes you can serialize a JSON containing parts of a deserialized object. They don't have to be in the same `JsonBuffer`, they just have to stay in memory during the whole operation, which is guarantied if you used scoped `JsonBuffer`s.

psigen, 2016-03-10:
(By the way, thanks!  I greatly appreciate the library and the opportunity to discuss this with you.)

I see your point, however, using a `const &` always has inherent risk of the container object going out of scope.  For example, the following code has the same problem, but is less "obvious":
```c++
JsonObject& my_object;
char line[512];

while (true) {
    line = readline();
    StaticJsonBuffer<200> jsonBuffer;
    jsonBuffer.parseObject(line);

    if (jsonBuffer['my_check'] == false)
        continue;
}
print(my_object['something'])  // Dangerous
```
In fact, this is exactly what a beginner might do if they were following your FAQ.  In general, if code is intended to be used by beginners, [Arduino recommends avoiding pass-by-reference altogether:](https://www.arduino.cc/en/Reference/APIStyleGuide)
> Don’t assume knowledge of pointers. Beginning users of C find this the biggest roadblock, and get very confused by & and *, so whenever you can avoid having them hanging out in the API, do so.

So, it seems that you are avoiding one confusing situation for beginners by creating another one, namely, that buffers are strange, write-only data structures that cannot be reused.  This is not a common semantic in any other data container: the `reset()` semantic is common in other standard containers such as:
* http://en.cppreference.com/w/cpp/memory/unique_ptr/reset
* http://en.cppreference.com/w/cpp/container/vector/clear
* http://en.cppreference.com/w/cpp/container/map/clear

bblanchon, 2016-03-10:
There seems to be several mistakes in the snippet. 
Where is the `my_object` reference set?

I know that Arduino recommends not using references and pointers, but what other solution do I have to make a zero-copy library?

The buffers of ArduinoJson are not write-only, read and write operations can be performed.
They are temporary / disposable / throw-aways, whatever you want to call them.

I'm not trying to copy the behavior of the STL containers.
They rely heavily on dynamic allocation.
That's why they are not suitable for embedded development.

psigen, 2016-03-10:
Oops, sorry, writing from an airport introduces syntax errors.

`const &` is definitely the right choice for a zero-copy library, but it means that regardless of how defensive you make the library, it is always possible to hold onto references while the parent goes out of scope.

I am not sure this justifies preventing people from using the data structure as a global, which is by far the most common memory semantic for Arduino users.  I would argue that asking users to understand C++ scoping semantics is more advanced than understanding that calling `reset()/clear()` will destroy all the previous references.

You are right, I didn't mean `write-only`, I meant `append-only`, as it is similar to append-only log semantics.

Most fixed-size solutions are avoiding this issue by passing a pure data structure as a pointer into a parsing call which then completely overwrites it.  I presented STL containers not because the memory management is acceptable, but because they are examples of containers that include member functions that mutate the data.

In any case, I now have a better understanding of why you went this route, so thanks very much!

bblanchon, 2016-03-11:
All the points you're making here are good.

To be fair, I'm not preventing people to use global variables, I just discourage them.

As a general rule, I'll try to encourage people to write what *I* consider good practice: 

1. no alloc
2. no copy
3. no global
4. RAII

See [FAQ: What's the best way to use the library?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#whats-the-best-way-to-use-the-library).

I was very pleased to answer your questions.
It's always good to review fundamental design decisions and see if they still make sense.
Thanks.

closed 2016-03-14 by psigen