bblanchon/ArduinoJson #244: Can't decode single char

danielebuso, 2016-03-15:
Hello, I'm trying to decode a string (received from the serial interface) but I can't decode two char from my object. Here is my code:
```c
struct payload_t {
  uint8_t node_id = nodeID;
  char param1;
  char param2;
  char data[16];
};
void serialEvent() {

  // Process with gatewayIn
  gatewayIn();
  
}
void gatewayIn() {
  
  // Prepare Buffer
  char buffer[200];
  getLine(buffer);

  // Prepare JSON object
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(buffer);

  // Convert json to payload
  payload_t data;
  data.node_id = 0;
  data.param1 = root["param1"];
  data.param2 = root["param2"];
  strcpy(data.data, root["data"]);
  Serial.println(data.node_id);
  Serial.println(data.param1);
  Serial.println(data.param2);
  Serial.println(data.data);

  // Send Data in the network
  meshSend(data, root["node_id"]);
  
}
void getLine(char * buffer) {
  
  uint8_t idx = 0;
  char c;
  do {
    while (Serial.available() == 0) ; // wait for a char this causes the blocking
    c = Serial.read();
    buffer[idx++] = c;
  } while (c != '\n' && c != '\r'); 
  buffer[idx] = 0;
  
}
```
Thanks in advance for any help

bblanchon, 2016-03-15:
Hello Daniele,

Can you please show the JSON input and the actual output of your program?

danielebuso, 2016-03-15:
Thanks for the answer, here is the JSON
```JSON
{"node_id":1,"param1":"a","param2":"b","data":"test"}
```
I can get the `node_id` and the `data` but i can't get the `param1` and `param2` chars

bblanchon, 2016-03-15:
This is because you're trying to convert a string to a char.

You can work around this by getting only the first char of the string:

```c++
data.param1 = root["param1"].as<const char*>()[0];
data.param2 = root["param2"].as<const char*>()[0];
```

danielebuso, 2016-03-15:
Thanks a lot, later I'll try

bblanchon, 2016-03-15:
You're welcome Daniele, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add you projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-03-15 by bblanchon

danielebuso, 2016-03-15:
It works perfectly, thank you again also for the fast support!