bblanchon/ArduinoJson #245: Can't parse part of JSON

dlevans, 2016-03-18:
Here is my JSON:
`{
coord: {
lon: -94.67,
lat: 38.98
},
weather: [
{
id: 802,
main: "Clouds",
description: "scattered clouds",
icon: "03n"
}
],
base: "stations",
main: {
temp: 284.32,
pressure: 1015,
humidity: 26,
temp_min: 280.15,
temp_max: 286.15
},
visibility: 16093,
wind: {
speed: 3.6,
deg: 360
},
clouds: {
all: 40
},
dt: 1458262440,
sys: {
type: 1,
id: 1647,
message: 0.0293,
country: "US",
sunrise: 1458303857,
sunset: 1458347358
},
id: 4276873,
name: "Overland Park",
cod: 200
}`

I can parse out everything except what's in here:

`weather: [
{
id: 802,
main: "Clouds",
description: "scattered clouds",
icon: "03n"
}
],`

I've tried ["weather"]["id"], ["weather"]["id"][0], ["weather"][0], etc but can't get anything out of that section. I've got everything else working, though. 

Any ideas? Thanks for the cool lib. Really helpful.

bblanchon, 2016-03-18:
Hello Dustin,

It looks like `root["weather"][0]["id"]` is the answer.

BTW, this is not valid/standard JSON, you're suppose to have quotation mark arounds object keys.
ArduinoJson normally reads this kind of input fine, but many tools don't.

Please read:

* [FAQ: Parsing succeeds but I can't read the values!](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#parsing-succeeds-but-i-cant-read-the-values)

dlevans, 2016-03-18:
Ahh, thanks. I'll give that a try. It looked strange to me, too. It's from http://openweathermap.org/api maybe I'll pick a different weather site's API.

bblanchon, 2016-03-18:
It's probably just the way you copy/paste the data to GitHub that made the quotation marks disappear.

dlevans, 2016-03-21:
Probably. I got it working with the [0]["id"] though. I had that part backwards. Thanks again for your help.

bblanchon, 2016-03-21:
You're welcome Dustin, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add you projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-03-21 by bblanchon