bblanchon/ArduinoJson #246: Assinnig to the variable the parsed JSON array

houbig, 2016-03-21:
Hi, first of all I would like to Thank you for this library. I'm a PHP programmer and use the arrays in all my projects. Your library helps me in my project for Arduino Server. I'm a new in C++ and I have an question to you. Can you help me to solve the issue with assigning to the variable the parsed JSON array, for future looping? In Arduino console I'm getting the error: _cannot convert 'Arduino Json::Json Array' to 'int*' in assignment_. The issue for the variables **led **and **cnt **

Here is my sketch.

`#include <ArduinoJson.h>
long id;
int* led;
int* cnt;

void setup () {
 Serial.begin(57600);
 while (!Serial) {}

 jsonParse("{\"id\":1351824120,\"data\":[48,2],\"dataE\":[10,20]}");
 runIt();
}

void loop () {

}

void runIt(){
  
  for (int i = 0; i < (sizeof(led)/sizeof(int)); i++) {
    Serial.print("LED> "); Serial.println(led[i]);
      for(int q = 0; q < cnt[i]; q++){
         Serial.print("Count > "); Serial.println(q+1);
      } 
          
  }  
  
}

void jsonParse(char json[]){
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(json);

  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }
  
  id = root["id"];
  JsonArray& ledArray = root["data"].asArray();
  JsonArray& countArray = root["dataE"].asArray();

  led = ledArray;
  cnt = countArray;

  ledArray.printTo(Serial);
  Serial.println();
  countArray.printTo(Serial);

  jsonBuffer = StaticJsonBuffer<200>(); 
}
`

bblanchon, 2016-03-21:
Hi,

If i'm not mistaken, it seems you want convert a `JsonArray&` to a C array.

Here is the one way to do it:

```c++
#define MAX_LED_COUNT 10

int led[MAX_LED_COUNT];
int ledCount = 0;

JsonArray& ledArray = root["data"].asArray();
for(int i : ledArray) { // <- range for, requires C++11
    led[ledCount++] = i;
    if (ledCount >= MAX_LED_COUNT) break;
}
```

houbig, 2016-03-22:
Hi, thank you! 

I thought that .asArray() is do the trick.

bblanchon, 2016-03-22:
You're welcome @houbig, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add you projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).
As you'll see, I added your example in the [Bag of Tricks section](https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks#extract-an-array-of-integer).

Also, a GitHub star is always appreciated :wink:

closed 2016-03-22 by bblanchon

houbig, 2016-03-22:
Done ;)

gutierrezps, 2016-06-07:
@bblanchon Could you please add .asArray() on API page? I couldn't find it there, and I would like to know which type of array it returns.

I tried to use it like this:

```
unsigned char leds[4];
root["leds"].asArray().copyTo(leds);
```

But it didn't work.

bblanchon, 2016-06-08:
@gutierrezps You may have found a bug! :bug:
As it's not related to #246, please [open a new issue](https://github.com/bblanchon/ArduinoJson/issues/new).
Also, please be more specific: "it didn't work" is very vague :-(