bblanchon/ArduinoJson #247: ESP8266 restart after upload "jsonParserExample"

wsun0928, 2016-03-22:
I not sure if this is the right place to ask. After I uploaded "jsonParserExample" to the Huzza(esp8266-12), I got "ets Jan  8 2013,rst cause:4, boot mode:(1,7)" from the monitor. Does anyone know what's going on?

bblanchon, 2016-03-22:
Yes, it's the right place to ask ;-)

Two questions:

1. Do have the same issue with other examples, in particular with JsonGeneratorExample?
2. Do you have an up-to-date Arduino core for ESP8266?

Also, please also have a look at:

* [Comparibility issues: ESP8266](https://github.com/bblanchon/ArduinoJson/wiki/Compatibility%20issues#esp8266)

wsun0928, 2016-03-23:
For JsonGeneratorExample. it works by returning the following:
{"sensor":"gps","time":1351824120,"data":[48.756081,2.302038]}
{
  "sensor": "gps",
  "time": 1351824120,
  "data": [
    48.756081,
    2.302038
  ]
}

And the JsonParserExample still have the same issue. 

Another question here, before I even parse the JSON, how do I retrieve the JSON from a web page?
I have a server and a php file at: http://10.0.0.22/mode.php

I get    ` [{"Mode":"Auto","Value":"1"}] `   in the browser, but I cannot figure out how to turn this into {\"Mode\":\"Auto\",\"Value\":\"1\"} in order to parse it.

bblanchon, 2016-03-23:
Your issue really looks like #101.
Are you sure that the Arduino Core is up to date?

To perform HTTP requests, have a look at [EthernetClient](https://www.arduino.cc/en/Reference/EthernetClient)

wsun0928, 2016-03-23:
right now I can read the test from the server but the parsing keeps failing...

I not sure what is the difference between the json I read form the server and the json I hardcode in the Arduino program.

`#include <ESP8266WiFi.h>
#include <ArduinoJson.h>

const char* ssid     = "Hello Kitty";
const char* password = "kiikiimeowmeow";

const char* host     = "10.0.0.22"; // Your domain
String path          = "/mode.php";
String path1         = "/fan.php";
String path2         = "/threshholdLuxJSON.php";
const int LedPin        = 2;
const int LuxPin        = A0;
int var=0;
void setup() {
 /*   pinMode(LedPin,OUTPUT);
    pinMode(LedPin,HIGH);
    pinMode(LuxPin,INPUT); */
    Serial.begin(115200);

    delay(10);
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);
    int wifi_ctr = 0;
    while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("WiFi connected");
  Serial.println("IP address: " + WiFi.localIP());

}

void loop() {
/*  int lux = analogRead(A0);
  Serial.println(lux);
  digitalWrite(LedPin,HIGH); */
  Serial.print("connecting to ");
  Serial.println(host);
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) 
  {
    Serial.println("connection failed");
    return;
  }else{
    Serial.println("Connection successful!!");
    }

  client.print(String("GET ") + path + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Connection: keep-alive\r\n\r\n");

  delay(500); // wait for server to respond

 // read response
  String section="header";
  while(client.available()){
    String line = client.readStringUntil('\r');
    // Serial.print(line);
    // we’ll parse the HTML body here
    Serial.println("Line");
      Serial.print(line);
  
      if (section=="header") { // headers..

      if (line.indexOf('[{') >=0) { // skips the empty space at the beginning 
        section="json";
        Serial.println("Found JSON");

        Serial.println(line);
        line = client.readStringUntil('\r');
        Serial.println(line);
    String line2 = line;
          StaticJsonBuffer<2000> jsonBuffer;
          line2.replace("[", "");
          Serial.println("Line2");
          
          Serial.println(line2);
         /* Serial.println(line);
          line.replace(']', ' ');
          line.trim();*/

          Serial.println(line);
        JsonObject& json_parsed = jsonBuffer.parseObject(line2);
        if (!json_parsed.success())
        {

            String json = "[{\"Mode\":\"Auto\",\"Value\":\"0\"}]";
            json.replace("[", "");
          JsonObject& json_parsed = jsonBuffer.parseObject(json);
         if (json_parsed.success())
        {
                  Serial.println("parseObject(json) Success!!");

        }

        
          Serial.println("parseObject(line) failed");
          return;
        }
        
         const char* device = json_parsed["Device"];
          int      mode = json_parsed["Mode"];

        Serial.println(device);
        Serial.println(mode);
          
      }
    }
         
}

   delay(50000); // wait for server to respond

}
`
Output the following
`Connecting to Hello Kitty
........WiFi connected

connecting to 10.0.0.22
Connection successful!!
Line
HTTP/1.1 200 OKLine

Date: Wed, 23 Mar 2016 19:09:57 GMTLine

Server: Apache/2.2.22 (Debian)Line

X-Powered-By: PHP/5.4.45-0+deb7u2Line

Vary: Accept-EncodingLine

Content-Length: 30Line

Keep-Alive: timeout=5, max=100Line

Connection: Keep-AliveLine

Content-Type: text/htmlLine

Line

 [{"Mode":"Auto","Value":"0"}]Found JSON

 [{"Mode":"Auto","Value":"0"}]

Line2

parseObject(json) Success!!
parseObject(line) failed`

wsun0928, 2016-03-24:
I read the #101, and check the lib. It seems to be fixed already.
`  char *end;`
 ` errno = 0;`
 ` strtol(_content.asString, &end, 10);`
And now I am getting
`Exception (3):`
`epc1=0x401003c1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4006f659 depc=0x00000000`
` `
`ctx: cont `
`sp: 3ffef470 end: 3ffefee0 offset: 01a0`

and a bunch of stack like shown in #101 
then
 `ets Jan  8 2013,rst cause:2, boot mode:(1,7)`
` ets Jan  8 2013,rst cause:4, boot mode:(1,7)`
`wdt reset`

bblanchon, 2016-03-28:
Any progress?
Did you manage to updated the Arduino Core for ESP8266?
Did you try to call `strtol(buffer, NULL, size(buffer))` to check if you have a buggy Core installed?

wsun0928, 2016-03-28:
Well, I am not sure how to explain this. After I upgraded arduino to the newer version, using external power source, the problem is gone...

bblanchon, 2016-03-28:
Well you just had a buggy core installed.
Thanks for the feedback.

closed 2016-03-28 by bblanchon