bblanchon/ArduinoJson #248: Array of JsonObject

EduardUA, 2016-03-23:
How to create a dynamic array of objects?

I use in such way: 

```
JsonObject &object_1 = jsonBuffer.createObject();
JsonObject &object_2 = jsonBuffer.createObject();
JsonObject &object_3 = jsonBuffer.createObject();
...
JsonObject &object_n = jsonBuffer.createObject();
```
But I need the other way, for example:

```
for(int i; i<n; i++)
{
object[i] = jsonBuffer.createObject();
}
```
Thanks.

bblanchon, 2016-03-23:
If you want to create a `JsonArray` of `JsonObject`, here is the way:

```c++
JsonArray& array = jsonBuffer.createArray();
for(int i; i<n; i++)
{
    JsonObject& obj = array.createNestedArray();
    obj["key1"] = "value_k1";
    obj["key2"] = "value_k2";
    obj["key3"] = "value_k3";
}
```

If you want to create a C array of `JsonObject`, you have to store pointers, not references:

```c++
#define MAX_OBJECT_COUNT 10

JsonObject* array[MAX_OBJECT_COUNT]

for(int i; i<n && i<MAX_OBJECT_COUNT; i++)
{
    JsonObject& obj = jsonBuffer.createObject();
    obj["key1"] = "value_k1";
    obj["key2"] = "value_k2";
    obj["key3"] = "value_k3";
    array[i] = &obj;
}
```

morphogencc, 2016-03-23:
I know you just closed this, but I'm having trouble creating an array of objects also, so I thought to put it in the same thread.

My code is:
```

JsonObject& buildJson(JsonBuffer& jsonBuffer) {
  JsonObject& root = jsonBuffer.createObject();
  root["selector_type"] = "part";
  JsonArray& touch_array = root.createNestedArray("data");

  for (int i = 0; i < 12; i++) {
    JsonObject& part = touch_array.createNestedObject();
    part["id"] = i;
    part["touched"] = false;
  }
  
  root.prettyPrintTo(Serial);
  return root;
}
```

Which I expect to return an object with an array of 12 items.

instead, the code returns:

```
{
  "selector_type": "part",
  "data": [
    {
      "id": 0,
      "touched": false
    },
    {
      "id": 1,
      "touched": false
    },
    {
      "id": 2,
      "touched": false
    },
    {
      "id": 3,
      "touched": false
    },
    {
      "id": 4,
      "touched": false
    },
    {}
  ]
}
```
Where it somehow stops after 4 items.  Any idea on why I can't produce an array of 12 objects?

closed 2016-03-24 by EduardUA