bblanchon/ArduinoJson #249: Json with RF 433Mhz

gustavogini, 2016-03-25:
Hello Benoit,

Is it possible use json to send a char using virtual wire library?

I`m creating the json char but how can i send it with virtual wire?

```
StaticJsonBuffer<200>`

 jsonBuffer;
JsonObject&` root = jsonBuffer.createObject();

root["Pan"] = jspan;
Serial.println(jspan);
root["Tilt"] = jstilt;
Serial.println(jstilt);
root["Zoom"] = jszoom;
Serial.println(jszoom);

root.printTo(??????);
    
    const char *msg = ??????;

    digitalWrite(led_pin, true); // Flash a light to show transmitting
    vw_send((uint8_t *)msg, strlen(msg));
    vw_wait_tx(); // Wait until the whole message is gone
    digitalWrite(led_pin, false);
    delay(200);
```

Thanks in advance!

Best Regards

bblanchon, 2016-03-26:
Hi,

Your program looks OK.
What's wrong with it?

There is a small improvement that you can do.
`JsonObject::printTo()` returns the number of written chars, so you can save a call to `strlen()`.

Regards,

gustavogini, 2016-03-26:
Hello Benoit!

I just change the `myJsonObject.printTo`  to `root.printTo` and it works very well.

Regards.

closed 2016-03-26 by gustavogini