bblanchon/ArduinoJson #250: Problem to Parse - Part 2

gustavogini, 2016-03-27:
Hello Benoit,
 
I am having problems to parse again, but now I am working with RF433MHz.
I suppose that problem is happening becouse I`m not converting correctly. Let me show you:

```
for (i = 0; i < buflen; i++)
	{
	    //Serial.print(buf[i], HEX);
            Serial.print(buf[i]);    
	    Serial.print(" ");          ***the information in the serial Monitor is: 123 34 80 97 110 34 58 34 49 54 48 48 34 44 34 84 105 108 116 34 58 34 49 53 48 48 34 44 34 90 111 111 109 34 58 34 49 53 48 48 34 125 parseObject() failed

	}
        char teste[256];              ***How can I convert the information to be parse very well?
        teste[256] = (char) buf[i];
        StaticJsonBuffer<256> jsonBuffer;
     JsonObject& root = jsonBuffer.parseObject(teste);
  	
  	if (!root.success()){
      Serial.println("parseObject() failed");
      return;
     }
     
      int panjson = root["Pan"];
      int tiltjson = root["Tilt"];
      int zoomjson = root["Zoom"];
```
Thanks

Regards

hemantsangwan, 2016-03-27:
```

char teste[256];              //<<<<<<< teste declaration should be here
for (i = 0; i < buflen; i++)
    {
        teste[i] = (char) buf[i];  //<<<<< Set buf[i] to teste[i] here
        //Serial.print(buf[i], HEX);
            Serial.print(buf[i]);    
        Serial.print(" ");          ***the information in the serial Monitor is: 123 34 80 97 110 34 58 34 49 54 48 48 34 44 34 84 105 108 116 34 58 34 49 53 48 48 34 44 34 90 111 111 109 34 58 34 49 53 48 48 34 125 parseObject() failed

    }
        StaticJsonBuffer<256> jsonBuffer;
     JsonObject& root = jsonBuffer.parseObject(teste);

    if (!root.success()){
      Serial.println("parseObject() failed");
      return;
     }

      int panjson = root["Pan"];
      int tiltjson = root["Tilt"];
      int zoomjson = root["Zoom"];
```

I'm sorry for being rude but users should ask/raise issues related to ArduinoJson. ( I have to say this because resolving such issues will slow down the developer of such an awesome codework)

gustavogini, 2016-03-27:
Thanks hemantsagwan.

closed 2016-03-27 by gustavogini

bblanchon, 2016-03-28:
Thank you very much @hemantsangwan.