bblanchon/ArduinoJson #251: parsing array of objects of multidimensional arrays

mckennatim, 2016-03-31:
Not sure if this type of JSON will work. If each object was just an array it will work but not if each object references a multidimensional array. 

    char* k = "[{\"temp0\":[[6,12,68],[8,20,57]]},{\"temp1\":[[6,0,67],[18,0,68],[21,30,58]]}]";
    StaticJsonBuffer<200> jsonBuffer;
    JsonArray& array = jsonBuffer.parseArray(k);
    array.prettyPrintTo(Serial); //[]
    Serial.println(array.size()); //0   
    JsonObject& temp0 = array[0]["temp0"];
    temp0.prettyPrintTo(Serial); //{}

btw: I love ArduinoJson.

bblanchon, 2016-04-01:
Hi Timothy,

Please try:

```c++
char k[] = "[{\"temp0\":[[6,12,68],[8,20,57]]},{\"temp1\":[[6,0,67],[18,0,68],[21,30,58]]}]";

DynamicJsonBuffer jsonBuffer;
JsonArray& array = jsonBuffer.parseArray(k);
JsonArray& temp0 = array[0]["temp0"];
JsonArray& temp1 = array[0]["temp1"];
```

As you can see, I changed three things:

1. `k` is now a `char[]` instead of a `char*` pointing to a `const char*` (you probably have a compiler warning for that).
2. I switched to `DynamicJsonBuffer`, in case 200 bytes was too small. You can switch back to `StaticJsonBuffer`, once the diagnostic is made.
3. I used a `JsonArray` for `temp0`, because it's an array (`[[6,12,68],[8,20,57]]`)

Please keep me informed with the result.

mckennatim, 2016-04-01:
That's great, thanks. Everything `.prettyPrintTo(Serial)` which is a good sign. Now I'll set to turning the `JsonArray` to an int[]... then get iterators to work for tempN sensors. 

You've built a great product with way above average wiki and issues sections so I am pretty confident.

bblanchon, 2016-04-01:
Converting `int[]` can be tricky.
Have a look at [Bag of Tricks: Extract and array of integer](https://github.com/bblanchon/ArduinoJson/wiki/Bag%20of%20Tricks#extract-an-array-of-integer)

mckennatim, 2016-04-01:
My checklist for ArduinoJson
- make sure the json string is good and working in outside prog
- remember, you pass in a pointer to a char[] in deserialize. If you want to test inside the function use char[] for your test json
- use `DynamicJsonBuffer` until everything checks out 
- `prettyPrintTo(Serial)` is your friend, use it right away at the first parse

My test code below is full of all these uneccessary things that helped me understand this.

      int t0[4][3];
      int t1[4][3];

      bool Sched::deseriProgs(char* kstr){
        StaticJsonBuffer<600> jsonBuffer2;
        JsonArray& root = jsonBuffer2.parseArray(kstr);
        JsonArray& temp0 = root[0]["temp0"];
        JsonArray& temp1 = root[1]["temp1"];
        for(int i = 0; i<4; i++){
          for (int j = 0; j<3; j++){
            t0[i][j]=temp0[i][j];
          }
        }
        for(int i = 0; i<4; i++){
          for (int j = 0; j<3; j++){
            t1[i][j]=temp1[i][j];
          }
        }
        return root.success();
      }

      bool Sched::deseriProgs(char* kstr){
        Serial.println("in deseriprogs");
        char k[] = "[{\"temp0\":[[6,12,68],[8,20,57]]},{\"temp1\":[[6,0,67],[18,0,68],[21,30,58]]}]";
        StaticJsonBuffer<600> jsonBuffer2;
        //DynamicJsonBuffer jsonBuffer2;
        JsonArray& root = jsonBuffer2.parseArray(kstr);
        root.prettyPrintTo(Serial);
        JsonArray& temp0 = root[0]["temp0"];
        temp0.prettyPrintTo(Serial);
        JsonArray& temp1 = root[1]["temp1"];
        temp0[1][2].prettyPrintTo(Serial);
        Serial.print("the # of 3tuples is: ");
        Serial.println(temp0[0].size());
        Serial.println("temp0");  
        for(int i = 0; i<4; i++){
          Serial.print("[");
          for (int j = 0; j<3; j++){
            t0[i][j]=temp0[i][j];
            Serial.print(t0[i][j]);
            Serial.print(",");
          }
          Serial.print("]");
          Serial.println("");
        }
        Serial.println("temp1");  
        for(int i = 0; i<4; i++){
          Serial.print("[");
          for (int j = 0; j<3; j++){
            t1[i][j]=temp1[i][j];
            Serial.print(t1[i][j]);
            Serial.print(",");
          }
          Serial.print("]");
          Serial.println("");
        }
        return 1;
      }

      temp0
      [6,12,68,]
      [8,20,57,]
      [22,0,68,]
      [23,30,58,]
      temp1
      [6,0,67,]
      [9,20,57,]
      [18,0,68,]
      [21,30,58,]

bblanchon, 2016-04-04:
Hi Timothy,

I strongly recommend to use iterators to extract the values from the array because it's much faster.
Unfortunately, when I tried to write the code to show you, I realized that it was not an easy task.
So I opened a new issue, see #254.

Can I close the current issue?

mckennatim, 2016-04-04:
sure

closed 2016-04-04 by mckennatim