bblanchon/ArduinoJson #252: Nested array in a nested array

gibix, 2016-03-31:
Hello,
I need to encode a json that has a value that is an array of arrays, like this:
`{"value1": "x", "value2": [["Yes","No"],["Maybe"]}`
is there a way to create a nested array as a value of a nested array in a json object?

I also tried to pass all the string `"[[\"Yes\",\"No\"],[\"Maybe\"]"` as a value, but the json library encodes the value also with the backslash!

Thanks for the support

bblanchon, 2016-04-01:
Hi Gilberto,

The canonical way to do this is:

```c++
JsonObject& root = jsonBuffer.createObject();

root["value1"] = "x";
JsonArray& value2 = root.createNestedArray("value2");

JsonArray& yesno = value2.createNestedArray();
yesno.add("Yes");
yesno.add("No");

JsonArray& maybe = value2.createNestedArray();
maybe.add("Maybe");
```

It's also possible to create the arrays and then put them in the object:

```c++
JsonObject& root = jsonBuffer.createObject();
JsonObject& value2 = jsonBuffer.createArray();
JsonObject& yesno = jsonBuffer.createArray();
JsonObject& maybe = jsonBuffer.createArray();

root["value1"] = "x";
root["value2"] = value2;
value2.add(yesno);
yesno.add("Yes");
yesno.add("No");
value2.add(maybe);
maybe.add("Maybe");
```

But it's a little less efficient in term of CPU and memory.

Lastly, it was possible to use the workaround with an undocumented feature:

```c++
JsonObject& root = jsonBuffer.createObject();
root["value1"] = "x";
root["value2"] = ArduinoJson::Internals::Unparsed("[[\"Yes\",\"No\"],[\"Maybe\"]");
```

BTW, I see that you are in the Arduino organisation.
Can you help me have [this pull request](https://github.com/arduino/Arduino/pull/4667) merged?
This is very important for me, because it'd allow to fix #170 without increasing the code size.
Thanks.

Regards,

hemantsangwan, 2016-04-01:
It should be added to documentation as it seems a common problem.

bblanchon, 2016-04-01:
@hemantsangwan It's now in the wiki [Bag of tricks: nested array in nested array](https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks#nested-array-in-a-nested-array)

gibix, 2016-04-01:
Thanks for the fast support!
 @bblanchon your PR will be tested later today and merged as soon as possible.

bblanchon, 2016-04-04:
@gibix Did you have an opportunity to look at my pull request ?

closed 2016-04-07 by bblanchon