bblanchon/ArduinoJson #255: "required from here" error

rvalle, 2016-04-04:
Hi!

I am trying to use ArduinoJson from AtmelStudio with for an ATmega 328p.

There is a functionality now: "Create Project from arduino sketch" which will enable to use proper debugging. I have tried with a couple of sketches and seems to work.

When trying to compile ArduinoJson  I am getting a lots of "required from here" error. The error is not very descriptive, so I have no idea were to even start looking.

They are been triggered from JsonObject.ipp.

Have anybody encountered this error before? Any idea what it could be related to? 

Thanks!

bblanchon, 2016-04-04:
Hi Raphael,

It would be very helpful to copy/paste the compiler log.

sammys, 2016-04-05:
Arduino: 1.6.8 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.hpp:153:0,
                 from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonArray.ipp:11,
                 from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonArray.hpp:181,
                 from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:9,
                 from ../Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from sketch/config.cpp:4:
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp: In instantiation of 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]':
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp:54:36:   required from 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TKey = const char*]'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:45:35:   required from 'const T ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
sketch/config.cpp:13:14:   required from here
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp:24:15: error: no matching function for call to 'ArduinoJson::JsonVariant::as()'
   return node ? node->content.value.as<T>() : JsonVariant::invalid<T>();
               ^
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp:24:15: note: candidates are:
In file included from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:15:0,
                 from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:10,
                 from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:8,
                 from ../Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from sketch/config.cpp:4:
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp:97:3: note: template<class T> const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsIntegral<T>::value, T>::type ArduinoJson::JsonVariant::as() const
   as() const {
   ^
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp:97:3: note:   template argument deduction/substitution failed:
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp: In substitution of 'template<class T> const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsIntegral<T>::value, T>::type ArduinoJson::JsonVariant::as() const [with T = char*]':
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp:24:15:   required from 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp:54:36:   required from 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TKey = const char*]'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:45:35:   required from 'const T ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
sketch/config.cpp:13:14:   required from here
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp:97:3: error: no type named 'type' in 'struct ArduinoJson::TypeTraits::EnableIf<false, char*>'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp: In instantiation of 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]':
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp:54:36:   required from 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TKey = const char*]'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:45:35:   required from 'const T ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
sketch/config.cpp:13:14:   required from here
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp:105:3: note: template<class T> const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsFloatingPoint<T>::value, T>::type ArduinoJson::JsonVariant::as() const
   as() const {
   ^
[... cut ...]
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
sketch/config.cpp:13:14:   required from here
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp:153:3: error: no type named 'type' in 'struct ArduinoJson::TypeTraits::EnableIf<false, ArduinoJson::JsonObject&>'
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

rvalle, 2016-04-05:
Hi, I found a thread on Stackoverlow that helped me with this issue:

https://stackoverflow.com/questions/15644938/what-does-in-instantiation-of-required-from-here-mean

It is actually not an error (as reported by AtmelStudio) but part of a multi-line warning.

The full warning is formed with the previous and next line, and it is related to unused parameters.

As a workaround, you can use -w (supress all warnings) in atmel studio, and this (with many other warnings) will disappear. This is actually what  Arduino IDE does.

There is probably another, thinner grain workaround in which you turn off only that kind of warning (unused parameters)... I havent yet tested it but maybe -Wunused-parameters would be the one.

I am not sure if this is a avr-g++ bug (for reporting error instead of warning) or an AtmelStudio bug for wrongly parsing the output of avr-g++. Either of them should be informed with the bug.

Rafael

bblanchon, 2016-04-05:
@rvalle: Please copy/paste the compiler log so I can fix this warning.

@sammys: Please copy/paste line 13 of `sketch/config.cpp`.

bblanchon, 2016-04-05:
@sammys: I opened a new issue for you, see #257. I moved your message there.

bblanchon, 2016-04-07:
@rvalle I'm closing this issue because there is nothing I can do without the compiler messages.
Please reopen if needed.

rvalle, 2016-04-07:
sorry @sammys, I have been super-busy these days:

http://pastebin.com/4Bph4TGM

Find the output in the pastebin.

bblanchon, 2016-04-07:
Thanks!

There is actually one warning:

```
warning: ignoring packed attribute because of unpacked non-POD field
```

but it's triggered at 7 locations:

```
DynamicStringBuilder.hpp(30,11):  'String& ArduinoJson::Internals::DynamicStringBuilder::_str'
JsonWriter.hpp(78,10):            'Print& ArduinoJson::Internals::JsonWriter::_sink'
Prettyfier.hpp(43,18):            'ArduinoJson::Internals::IndentedPrint& ArduinoJson::Internals::Prettyfier::_sink'
JsonArraySubscript.hpp(69,14):    'ArduinoJson::JsonArray& ArduinoJson::JsonArraySubscript::_array'
JsonObjectSubscript.hpp(79,15):   'ArduinoJson::JsonObject& ArduinoJson::JsonObjectSubscript<const char*>::_object'
JsonObjectSubscript.hpp(79,15):   'ArduinoJson::JsonObject& ArduinoJson::JsonObjectSubscript<const String&>::_object'
JsonObjectSubscript.hpp(80,8):    'const String& ArduinoJson::JsonObjectSubscript<const String&>::_key'
```

bblanchon, 2016-04-07:
This warning is due to the flag `--fpack-struct`

> -fpack-struct[=n]
>        Without a value specified, pack all structure members together
>        without holes. 
>        When a value is specified (which must be a small power of two),
>        pack structure members according to this value, representing
>        the maximum alignment (that is, objects with default alignment
>        requirements larger than this will be output potentially
>        unaligned at the next fitting location.
> 
>        Warning: the -fpack-struct switch causes GCC to generate code
>        that is not binary compatible with code generated without that
>        switch. 
>        Additionally, it makes the code suboptimal. Use it to conform
>        to a non-default application binary interface.

I'm not sure if I can prevent this warning...

rvalle, 2016-04-08:
I see, I think is a default Atmel Studio compilation flag.
I noticed that Arduino, indeed does not use it.
I will test taking it out.

rvalle, 2016-04-08:
yes, that flag is the responsible for the warning.
0 warnings without it.

bblanchon, 2016-04-08:
Thanks for the feedback.
I'll check if I can eliminate the warning when the flag is present.

bblanchon, 2016-08-23:
[This was added to the wiki](https://github.com/bblanchon/ArduinoJson/wiki/Compatibility-issues#gcc-warning-ignoring-packed-attribute-because-of-unpacked-non-pod-field)

closed 2016-08-23 by bblanchon