bblanchon/ArduinoJson #256: How to make a request to a server and receive the JSON response?

horendus, 2016-04-04:
HEY!

I just need to make a quick arduino sketch to do a POST request to http://ip.jsontest.com/ and parse the JSON response to my arduino 

This is what I get when I paste  http://ip.jsontest.com/ into my browser so I know the server works. 
{"ip": "34.33.111.135"}

None of the examples show an example of this!

It would be great for a quick example, I know this must be something easy.

bblanchon, 2016-04-04:
Hi Jasper,

You are right, that would be a very good candidate for the Wiki:
https://github.com/bblanchon/ArduinoJson/wiki/Examples

I'm not sure, I'll have time to do that.
Would you mind adding it yourself?

horendus, 2016-04-04:
If I manage to work it out I will!

bblanchon, 2016-04-06:
@horendus Please have a look at [this file](https://github.com/bblanchon/ArduinoJson/blob/98652448373d806efde8afad3687a72dba31b572/examples/JsonHttpClient/JsonHttpClient.ino)
If you'r satisfied, I'll include it the the offcial set of excamples.

Note that http://ip.jsontest.com/ what not available (error 503), so I used another service.

horendus, 2016-04-07:
Looks great many Thanks!

closed 2016-04-08 by bblanchon