bblanchon/ArduinoJson #257: Error with as<char*>()

bblanchon, 2016-04-05:
From @sammys:

> Arduino: 1.6.8 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
> 
> ```
> In file included from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.hpp:153:0,
                 > from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonArray.ipp:11,
                 > from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonArray.hpp:181,
                 > from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:9,
                 > from ../Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 > from sketch/config.cpp:4:
> ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp: In instantiation of 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]':
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp:54:36:   required from 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TKey = > const char*]'
> JsonVariantBase.hpp:45:35:   required from 'const T ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> sketch/config.cpp:13:14:   required from here
> ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp:24:15: error: no matching function for call to 'ArduinoJson::JsonVariant::as()'
   > return node ? node->content.value.as<T>() : JsonVariant::invalid<T>();
               > ^
> ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp:24:15: note: candidates are:
> In file included from JsonBuffer.hpp:15:0,
                 > from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:10,
                 > from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 > from ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:8,
                 > from ../Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 > from sketch/config.cpp:4:
> JsonVariant.hpp:97:3: note: template<class T> const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsIntegral<T>::value, T>::type ArduinoJson::JsonVariant::as() const
   > as() const {
   > ^
> JsonVariant.hpp:97:3: note:   template argument deduction/substitution failed:
JsonVariant.hpp: In substitution of 'template<class T> const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsIntegral<T>::value, T>::type ArduinoJson::JsonVariant::as() const > [with T = char*]':
> ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp:24:15:   required from 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp:54:36:   required from 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TKey = > const char*]'
> JsonVariantBase.hpp:45:35:   required from 'const T ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> sketch/config.cpp:13:14:   required from here
> JsonVariant.hpp:97:3: error: no type named 'type' in 'struct ArduinoJson::TypeTraits::EnableIf<false, char*>'
> ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp: In instantiation of 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]':
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp:54:36:   required from 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TKey = > const char*]'
> JsonVariantBase.hpp:45:35:   required from 'const T ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> sketch/config.cpp:13:14:   required from here
> JsonVariant.hpp:105:3: note: template<class T> const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsFloatingPoint<T>::value, T>::type ArduinoJson::JsonVariant::as() const
   > as() const {
   > ^
> JsonVariant.hpp:105:3: note:   template argument deduction/substitution failed:
JsonVariant.hpp: In substitution of 'template<class T> const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsFloatingPoint<T>::value, T>::type ArduinoJson::JsonVariant::as() > const [with T = char*]':
> ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp:24:15:   required from 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp:54:36:   required from 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TKey = > const char*]'
> JsonVariantBase.hpp:45:35:   required from 'const T ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> sketch/config.cpp:13:14:   required from here
> JsonVariant.hpp:105:3: error: no type named 'type' in 'struct ArduinoJson::TypeTraits::EnableIf<false, char*>'
> ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp: In instantiation of 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]':
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp:54:36:   required from 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TKey = > const char*]'
> JsonVariantBase.hpp:45:35:   required from 'const T ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> sketch/config.cpp:13:14:   required from here
> JsonVariant.hpp:112:3: note: template<class T> const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsSame<T, String>::value, T>::type ArduinoJson::JsonVariant::as() const
   > as() const {
   > ^
> JsonVariant.hpp:112:3: note:   template argument deduction/substitution failed:
JsonVariant.hpp: In substitution of 'template<class T> const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsSame<T, String>::value, T>::type ArduinoJson::JsonVariant::as() > const [with T = char*]':
> ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp:24:15:   required from 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]'
../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp:54:36:   required from 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TKey = > const char*]'
> JsonVariantBase.hpp:45:35:   required from 'const T ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> sketch/config.cpp:13:14:   required from here
> JsonVariant.hpp:112:3: error: no type named 'type' in 'struct ArduinoJson::TypeTraits::EnableIf<false, char*>'
> ../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp: In instantiation of 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]':
[... 53 lines cut ...]
   > as() const {
   > ^
> JsonVariant.hpp:153:3: note:   template argument deduction/substitution failed:
[... cut ...]
> JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
> sketch/config.cpp:13:14:   required from here
> JsonVariant.hpp:153:3: error: no type named 'type' in 'struct ArduinoJson::TypeTraits::EnableIf<false, ArduinoJson::JsonObject&>'
> exit status 1
> Error compiling for board Arduino/Genuino Mega or Mega 2560.
> ```

I beleive that the faulty line is

```c++
root["something"].as<char*>()
```

instead of

```c++
root["something"].as<const char*>()
```

sammys, 2016-04-05:
sketch/config.cpp for the branch you've committed in:

#include <SoftwareSerial.h>
//#include <FileIO.h>
#include <stdlib.h>
#include <ArduinoJson.h>
#include "config.h"

char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

Config::Config() {
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& configJson = jsonBuffer.parseObject(json);

  name = configJson["name"];
}

In file included from .../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.hpp:153:0,
                 from .../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonArray.ipp:11,
                 from .../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonArray.hpp:181,
                 from .../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:9,
                 from .../Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from sketch/config.cpp:4:
.../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp: In instantiation of 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = char*]':
.../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObjectSubscript.hpp:54:36:   required from 'TValue ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TKey = const char*]'
.../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:45:35:   required from 'const T ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
.../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:40:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
sketch/config.cpp:13:8:   required from here
.../Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonObject.ipp:24:71: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
   return node ? node->content.value.as<T>() : JsonVariant::invalid<T>();
                                                                       ^
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

bblanchon, 2016-04-05:
The problem here is the type of `name`.
It's currently a `char*` should be a `const char*`.

bblanchon, 2016-04-05:
I'm closing this issue because I merged the enhancement into the master branch.

Please comment or open a new issue if you still need assistance.

closed 2016-04-05 by bblanchon

sammys, 2016-04-06:
Yup. Awesome. Thanks!